openwrt/luci #5536: Can't login luci

4NobleTruths, 2021-11-21:
on Raspberry Pi 4 using imagebuilder-21.02.1-bcm27xx-bcm2711
can't login luci “Invalid username and/or password! Please try again.”
logread error: daemon.err uhttpd[3063]: luci: failed login on / for root from 192.168...
Same as this from forum, but those solutions don't work: https://forum.openwrt.org/t/unable-to-logon-via-luci-able-via-ssh/25408

jow-, 2021-11-21:
Works fine for me with standard images. Make sure your /etc/shadow uses MD5 hashes, delete any existing `sysauth` cookie from your browser.

closed 2021-11-21 by jow-

4NobleTruths, 2021-12-25:
Same problem on completely different device GL-e750 using image from imagebuilder-21.02.1-ath79-nand.
Was able to login luci before i set passwd in ssh console.
I didn't change type of shadow hashes, nor do I know how.

jow-, 2022-01-09:
You likely included something in your image builder invocation which causes problems.

4NobleTruths, 2022-01-11:
This  caused that:
https://bugs.openwrt.org/index.php?do=details&task_id=4044