openwrt/luci #5541: luci-mod-system reboot

5p0ng3b0b, 2021-11-22:
The module luci-app-advanced-reboot module has a shutdown button. Would it be possible (and practical) to move/add this function to the reboot menu?

jow-, 2021-12-03:
No, since there is no platform neutral way to offer this. Only a small subset of OpenWrt supported targets offers proper shutdown functionality and we do not want to maintain model lists in the code just to guard this feature.

closed 2021-12-03 by jow-

5p0ng3b0b, 2021-12-05:
No worries and thanks for looking into it. I can clean shut down via cli and only need it occasionally. Better to clean shutdown than pulling the plug running transmission server and then having to fsck a 4tb drive afterwards.