openwrt/luci #5555: add additional sysupgrade switches (-k -u) to Generate Archive button

shoulders, 2021-11-27:
I would like the additional (-k -u) switches added to the **Generate Archive** option.

[image]

These options would allow me to create a backup config file without unmodified files to allow better compatibility when I do a manually upgrade or restore my settings to another router and make it much easier too re-install all of the missing required packages because they will be listed in `/etc/backup/installed_packages.txt`
```
grep "\toverlay" /etc/backup/installed_packages.txt | cut -f1 | xargs -r opkg install
```
I can imagine it will look similar to the **Flash new firmware image**

[image]

Related: #5530 #4160

jospezial, 2022-02-26:
@Ansuel @ynezz @rmilecki @aparcar @jow- @hnyman 
I am also interested in more options from sysupgrade in luci for "Generate Archive"
I could make use of -u , -c and -o.

I could not find the file in luci source with the command ```sysupgrade -b``` or ```sysupgrade --create-backup```

On the device:
```
grep -r create-backup /www/
/www/cgi-bin/cgi-backup:--create-backup
/www/cgi-bin/cgi-download:--create-backup
/www/cgi-bin/cgi-exec:--create-backup
/www/cgi-bin/cgi-upload:--create-backup
```

jospezial, 2022-02-26:
See also:
https://github.com/openwrt/openwrt/issues/9348

jospezial, 2022-02-27:
@shoulders Another way is to modify /sbin/sysupgrade on the device (or package/base-files/files/sbin/sysupgrade in openwrt source)
```
# initialize defaults
export MTD_ARGS=""
export MTD_CONFIG_ARGS=""
export INTERACTIVE=0
export VERBOSE=1
export SAVE_CONFIG=1
export SAVE_OVERLAY=0
export SAVE_OVERLAY_PATH=
export SAVE_PARTITIONS=1
export SAVE_INSTALLED_PKGS=0
export SKIP_UNCHANGED=0
export CONF_IMAGE=
export CONF_BACKUP_LIST=0
export CONF_BACKUP=
export CONF_RESTORE=
export NEED_IMAGE=
export HELP=0
export FORCE=0
export TEST=0
export UMOUNT_ETCBACKUP_DIR=0
```
After editing ```export SKIP_UNCHANGED=0``` to ```export SKIP_UNCHANGED=1``` "Generate Archive" does skip from backup files that are equal to those in /rom.
Edit also ```export SAVE_INSTALLED_PKGS=0``` to ```export SAVE_INSTALLED_PKGS=1```

If you want to backup the script too then you have to add /sbin/sysupgrade into /etc/sysupgrade.conf because even -o does not catch it. But beware of that the sysupgrade script then never gets updates if you always restore it.

Ansuel, 2022-02-27:
What I still didn't understand is if these options are available from CMD
Or we should add supporto for them in sysupgrade first.