openwrt/luci #5559: luci-app-commands: Custom arguments not working with "/"

KristerLundmark, 2021-11-29:
When using "Custom arguments" to a created command the resulting parsed command removes custom arguments starting with slash ("/").

The commands works ok if putting the parameters in the command itself, but not as custom arguments.

Examples:

Command: ls /etc/config
=> Works ok

Command: ls
Custom parameter: /etc/config
=> Does not work, the resulting command is: "ls etc"

Command: ls
Custom parameter: /tmp
=> Does not work, the resulting command is: "ls tmp"

Command: ls
Custom parameter: /
=> Does not work, the resulting command is: "ls"

I have tried quoting the custom parameter when entering value in luci, but this doesn't work either, when using double- or single quoting, the parameter is removed:

Command: ls
Custom parameter: "/etc/config"
=> Does not work, the resulting command is: "ls"

Regards,
Krister

wulfy23, 2021-12-05:
```
%2Fetc%2Fconfig
```

KristerLundmark, 2021-12-06:
No, that doesn't work (ls as command, %2Fetc%2Fconfig as parameter value):

# ls etc
ls: etc: No such file or directory

wulfy23, 2021-12-07:
should have tested it first \0/

appears sanitized for good reason...

ttyd or similar are great if you want shell access... or look into libexec+acl.d for more control for integrated functions...

KristerLundmark, 2021-12-07:
Thats ok, I use ssh (putty) for shell access, but it's more convenient to use app-commands to e.g. quick check all my devices (I have a shell script that loops through all devices). It's not critical, but more like "good-to-have" to not have to create app-commands with hard-coded parameter values.

closed 2022-10-24 by jow-