openwrt/luci #5560: <adblock>:  [SyntaxError] unexpected token: string literal  in https://192.168.1.1/luci-static/resources/protocol/modemmanager.js?v=2.2.4:7   at compileClass/< (https://192.168.1.1/luci-static/resources/luci.js?v=git-21.314.39479-e1ccb66:181:29)

sla-te, 2021-11-29:
After activating adblock I am getting the exception mentioned in the subject.

I am running http://rpi4.wulfy23.info/builds/rpi-4_21.02.1_1.0.7-3_r16325_extra/rpi4.64-21.02.1-27006-1.0.7-3-r16325-ext4-sys.img.gz

**dmesg**
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.154 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16325-88151b8303)) #0 SMP Sun Oct 24 09:01:35 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45336 r8192 d32488 u86016
[    0.000000] pcpu-alloc: s45336 r8192 d32488 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:3B:09:72 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=07022009-02 rootfstype=squashfs,ext4 rootwait pci=pcie_bus_perf fsckparts
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 1837624K/2019328K available (7102K kernel code, 432K rwdata, 1144K rodata, 512K init, 541K bss, 116168K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from 0xffffffc0108a0b98 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000158] Console: colour dummy device 80x25
[    0.000394] printk: console [tty1] enabled
[    0.000429] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000450] pid_max: default: 32768 minimum: 301
[    0.000610] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000663] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001278] cgroup: Disabling memory control group subsystem
[    0.002081] ASID allocator initialised with 32768 entries
[    0.002174] rcu: Hierarchical SRCU implementation.
[    0.002819] smp: Bringing up secondary CPUs ...
[    0.003225] Detected PIPT I-cache on CPU1
[    0.003282] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003736] Detected PIPT I-cache on CPU2
[    0.003771] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004284] Detected PIPT I-cache on CPU3
[    0.004318] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004387] smp: Brought up 1 node, 4 CPUs
[    0.004414] SMP: Total of 4 processors activated.
[    0.004428] CPU features: detected: 32-bit EL0 Support
[    0.004441] CPU features: detected: CRC32 instructions
[    0.004496] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004514] CPU: All CPU(s) started at EL2
[    0.004543] alternatives: patching kernel code
[    0.012196] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.012237] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.013982] pinctrl core: initialized pinctrl subsystem
[    0.014864] NET: Registered protocol family 16
[    0.016300] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.016866] cpuidle: using governor ladder
[    0.016894] cpuidle: using governor menu
[... 340 lines cut ...]
[   22.167725] device wlan1 entered promiscuous mode
[   22.190786] 55-dummy> placeholder
[   22.239222] 71-macruneveryboot> /etc/custom/d22506ef2dd8.sh everyboot [ok]
[   22.289444] rc.custom> running mount-cmds: /bin/mountcmds.sh [mount]
[   22.342857] rc.local> no commit: nothing or nochanges
[   22.372788] rc.local> ############################################################## end rc.local
[   30.727274] br-bb6be9367275: port 1(veth0538829) entered blocking state
[   30.735546] br-bb6be9367275: port 1(veth0538829) entered disabled state
[   30.744260] device veth0538829 entered promiscuous mode
[   30.752027] br-bb6be9367275: port 1(veth0538829) entered blocking state
[   30.760162] br-bb6be9367275: port 1(veth0538829) entered forwarding state
[   30.768903] br-bb6be9367275: port 1(veth0538829) entered disabled state
[   31.295338] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   31.303544] br-lan: port 3(wlan1) entered blocking state
[   31.310378] br-lan: port 3(wlan1) entered forwarding state
[   33.094708] eth0: renamed from vethe146d09
[   33.129781] IPv6: ADDRCONF(NETDEV_CHANGE): veth0538829: link becomes ready
[   33.137986] br-bb6be9367275: port 1(veth0538829) entered blocking state
[   33.146072] br-bb6be9367275: port 1(veth0538829) entered forwarding state
[   33.154355] IPv6: ADDRCONF(NETDEV_CHANGE): br-bb6be9367275: link becomes ready
[   37.549738] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   37.557508] br-lan: port 2(wlan0) entered blocking state
[   37.564072] br-lan: port 2(wlan0) entered forwarding state
[   59.317185] EEE_DISABLE [off]
```

jow-, 2021-11-29:
Please open https://192.168.1.1/luci-static/resources/protocol/modemmanager.js in your browser and paste the exact contents here.

sla-te, 2021-11-29:
```js
'use strict';'require fs';'require form';'require network';function getModemList(){return fs.exec_direct('/usr/bin/mmcli',['-L']).then(function(res){var lines=(res||'').split(/\n/),tasks=[];for(var i=0;i<lines.length;i++){var m=lines[i].match(/\/Modem\/(\d+)/);if(m)
tasks.push(fs.exec_direct('/usr/bin/mmcli',['-m',m[1]]));}
return Promise.all(tasks).then(function(res){var modems=[];for(var i=0;i<res.length;i++){var man=res[i].match(/manufacturer: ([^\n]+)/),mod=res[i].match(/model: ([^\n]+)/),dev=res[i].match(/device: ([^\n]+)/);if(dev){modems.push({device:dev[1].trim(),manufacturer:(man?man[1].trim():'')||'?',model:(mod?mod[1].trim():'')||dev[1].trim()});}}
return modems;});});}
network.registerPatternVirtual(/^mobiledata-.+$/);network.registerErrorCode('MM_CONNECT_FAILED',_('Connection attempt failed.'));network.registerErrorCode('MM_DISCONNECT_IN_PROGRESS',_('Modem disconnection in progress. Please wait.'));network.registerErrorCode('MM_CONNECT_IN_PROGRESS',_('Modem connection in progress. Please wait. This process will timeout after 2 minutes.'));network.registerErrorCode('MM_TEARDOWN_IN_PROGRESS',_('Modem bearer teardown in progress.'));network.registerErrorCode('MM_MODEM_DISABLED',_('Modem is disabled.'));network.registerErrorCode('DEVICE_NOT_MANAGED',_('Device not managed by ModemManager.'));network.registerErrorCode('INVALID_BEARER_LIST',_('Invalid bearer list. Possibly too many bearers created.  This protocol supports one and only one bearer.'));network.registerErrorCode('UNKNOWN_METHOD',_('Unknown and unsupported connection method.'));network.registerErrorCode('DISCONNECT_FAILED',_('Disconnection attempt failed.'));return network.registerProtocol('modemmanager',{getI18n:function(){return _('ModemManager');},getIfname:function(){return this._ubus('l3_device')||'modemmanager-%s'.format(this.sid);},getOpkgPackage:function(){return'modemmanager';},isFloating:function(){return true;},isVirtual:function(){return true;},getDevices:function(){return null;},containsDevice:function(ifname){return(network.getIfnameOf(ifname)==this.getIfname());},renderFormOptions:function(s){var dev=this.getL3Device()||this.getDevice(),o;o=s.taboption('general',form.ListValue,'_modem_device',_('Modem device'));o.ucioption='device';o.rmempty=false;o.load=function(section_id){return getModemList().then(L.bind(function(devices){for(var i=0;i<devices.length;i++)
this.value(devices[i].device,'%s - %s'.format(devices[i].manufacturer,devices[i].model));return form.Value.prototype.load.apply(this,[section_id]);},this));};s.taboption('general',form.Value,'apn',_('APN'));s.taboption('general',form.Value,'pincode',_('PIN'));o=s.taboption('general',form.ListValue,'auth',_('Authentication Type'));o.value('both',_('PAP/CHAP (both)'));o.value('pap','PAP');o.value('chap','CHAP');o.value('none',_('None'));o.default='none';o=s.taboption('general',form.Value,'username',_('PAP/CHAP username'));o.depends('auth','pap');o.depends('auth','chap');o.depends('auth','both');o=s.taboption('general',form.Value,'password',_('PAP/CHAP password'));o.depends('auth','pap');o.depends('auth','chap');o.depends('auth','both');o.password=true;o=s.taboption('general',form.ListValue,'iptype',_('IP Type'));o.value('ipv4v6',_('IPv4/IPv6 (both - defaults to IPv4)'))
o.value('ipv4',_('IPv4 only'));o.value('ipv6',_('IPv6 only'));o.default='ipv4v6';o=s.taboption('advanced',form.Value,'mtu',_('Override MTU'));o.placeholder=dev?(dev.getMTU()||'1500'):'1500';o.datatype='max(9200)';o=s.taboption('general',form.Value,'signalrate',_('Signal Refresh Rate'),_("In seconds"));o.datatype='uinteger';}});
```

jow-, 2021-11-29:
What kind of "Adblock" do you mean btw.? Browser Addon? The "luci-app-adblock" package?

sla-te, 2021-11-29:
luci-adblock

jow-, 2021-11-29:
And the error is gone when you disable the adblock service?

sla-te, 2021-11-29:
I can only "suspend" the adblock service via luci and after doing that the error is still visible and even when it is "suspended", the error is being redisplayed upon refreshing the page.
[image]

jow-, 2021-11-29:
So what did you do exactly that caused the error to appear?

sla-te, 2021-11-29:
> So what did you do exactly that caused the error to appear?

Only navigating to https://192.168.1.1/cgi-bin/luci/admin/services/adblock/overview

jow-, 2021-11-29:
Does the error appear on other pages? E.g. in the network interface overview.

sla-te, 2021-11-29:
> Does the error appear on other pages? E.g. in the network interface overview.

No, it only appears at https://192.168.1.1/cgi-bin/luci/admin/services/adblock/overview

wulfy23, 2021-12-01:
i've luckily stumble upon this... jow's been very kind

due to the multiple hacks in my custom build... best first point of call is to ask/report most issues on the forum first...

that said... I installed your version, enabled and started dbus/modemmanager and do not get this error... so I probably would have been lost as to how to help mostly anyway...
(suspect the prescence of an actual modemmanaged interface is the differentiating factor here tho' still can't rule out build related issues )...

some things you can try;

```
opkg update; opkg install --force-reinstall modemmanager luci-proto-modemmanager; reboot
```
(test luci again)

bring down modemmanaged interface ( ifdown XYZ ) or stop modemmanger service and test again... ( apologies not familiar with modemmangar to provide specifics... )

install 21.02.0 (or r1806x) and test with this to see if the issue is present also there....

sla-te, 2021-12-30:
Sorry for the very late reply.

- It works fine with the bootstrap theme.
- Tried doing the commands above but that doesn't fix it for the argon theme

Meanwhile i have upgraded to "current" and the issue does not happen there anymore.

closed 2021-12-30 by sla-te