openwrt/luci #5562: luci: potential access lockout DoS

systemcrash, 2021-12-01:
When applying settings in LuCI which involve a change of IP address, or other form of access medium change with a potential lock out, LuCI waits for a re-connection before committing to the new settings. If there is no connection, it rolls back after the ~90 seconds is up. 

When surfing in via the new access IP, the settings are immediately accepted, *without an authenticated log in*

There exist potentially innocuous or malicious access on the (new) IP might trigger a commit.

I think that at a minimum, the settings should not be committed until (at least) an authenticated user has logged in.

systemcrash, 2021-12-01:
OpenWRT 21.02

jow-, 2021-12-01:
This decision was deliberate, reasoning can be seen in the original commit e5a1ac02289e8fde8ddbd05bbb21ac448c661ae3

IMHO the decision to change the LAN IP (or any other connection impacting setting) is a conscious one, so one should be prepared to reconnect on the new subnet anyway. I see little DoS potential in "prematurely" confirming the new settings. In cases where the operator does not trust the local environment and/or the rollback mechanism, one can chose to apply the settings immediately using "Apply unchecked" so a viable alternative exists.

Due to that, and due to the associated complexities and regression potential, I will not change the current logic.

closed 2021-12-01 by jow-

systemcrash, 2021-12-01:
OK - although if someone comes with a way of at least putting it behind authentication (someone that's logged in), would that be acceptable?

jow-, 2021-12-01:
I guess so. The original motivation for not putting it behind any authentication was to reduce the time to confirmation in order to lessen the likelihood of the rollback timer expiring. Since the timer has been extended to 90s, moving it behind authentication is probably okay