openwrt/luci #5575: SyntaxError Failed to execute 'open' on 'XMLHttpRequest': Invalid URL

n3gwg, 2021-12-03:
I am using Chrome Version 96.0.4664.55 (Official Build) (x86_64) on macOS Big Sur 11.6.1 and upon opening the main webpage of the router running OpenWRT I get the error as follows hereupon. Has anyone seen this occurring before and is there a recommended fix or workaround for it?

SyntaxError
Failed to execute 'open' on 'XMLHttpRequest': Invalid URL

Thanks!

Stuart

jow-, 2021-12-03:
Dupe of #5507

closed 2021-12-03 by jow-

n3gwg, 2021-12-03:
With all due respect, that issue was closed with not so much as even a single comment or concern about it being a bug or any troubleshooting. Might I ask why?

Moreover, a more microscopic and granular review of my bug filing speaks to this bug being present within a macOS environment, whereas the other report would seem to instantiate the environment being Microsoft Windows based. Whilst I am not saying the two are not the same or can't be merged, I am stating that the environments are clearly different betwixt the two reports.

n3gwg, 2021-12-03:
I am not disagreeing it is a duplicate. Rather the more salient and persistent question remains, as to why the original report was marked closed absent any discussion whatsoever on the issue or troubleshooting?

jow-, 2021-12-03:
> why the original report was marked closed absent any discussion whatsoever on the issue or troubleshooting?

Because it too was a duplicate of an much older report #3747 - I am simply cleaning up tickets.

n3gwg, 2021-12-03:
Fair enough. However, I would advise having referenced the original bug not a reference to a reference, etc...
Moreover, my question remains, #3747 also was closed absent any resolution or significant debugging / troubleshooting. Why was that bug closed absent a resolution then? You seemed to have closed that too. Is your position within the openwrt development community to just close bug reports even if they are not resolved? This really makes little sense.

jow-, 2021-12-04:
> #3747 also was closed

No? It is open.

n3gwg, 2021-12-04:
Ok, it looked closed to me, but I will add my comments to that report then also. Thank you. If I made a mistake and its open, then I sincerely apologize for my comments!