openwrt/luci #5579: Disabling MAC filter deletes the filters in wireless file. No chance to enanbnle the filters again

derrapf, 2021-12-04:
Once you define MAC filters for wireless (and that may be a lot), they are stored in the file "wireless" like
`	...
        list maclist '00:01:02:03:04:05'
	list maclist '06:07:08:09:AA:AB'
	list maclist 'AC:AD:AE:AF:B0:B2'
	...
`
Once you hit the option "disable" at "MAC Address filters" those entries in wireless are deleted.

So there is no chance to geht them back when you hit "enable" again.
You have to edit all the entries egain.
This is not only an error but max annoying.

BR
Ralf

closed 2021-12-09 by jow-

derrapf, 2021-12-10:
I downloaded  form.js and replaced  mine.
That does not solve the error.
When I now want to enable the former mac list again, all I get is a combo box with mac adresses,
Now I still have to manually add every single one again.
That is not conveniant.

What I expect , is that I can manage a list of MAC addresses and could disable/enable the whole list with one click.

BR
Ralf

jow-, 2021-12-10:
You need to update wireless.js as well.

derrapf, 2021-12-11:
Which one?
  modules/luci-mod-status/htdocs/luci-static/resources/view/status/wireless.js
or
  modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js  ?

BR Ralf

jow-, 2021-12-11:
The latter, see commit https://github.com/openwrt/luci/commit/c534f0671d9a7d0f12d38ece20fcfb3bc19509b2 which closed this issue.

derrapf, 2021-12-11:
I changed file /www/luci-static/resources/view/network/wireless.js with the one you linked abovec and  rebooted the device.
Unfortunatelly that did not change the bevaviour:

 * I added a list of MAC adresses; then I saved them.
 * After that I disabled MAC Filtering, saved again.
 * Then I enabled MAC-Filtering again, and the list was gone. 
 * All that is  left is a combobox where I can choose from 30 or more MAC addresses. 

That's the only change from the initial behaviour, where I had an empty combobox.
But I (probably nobody) do not want to create the MAC-list from scratch, once I disabled it. 
The combox with the MAC Adresses does not really help.
In this case it's more practical to do a backup from /etc/config/wireless instead of adding all those MAC addresse manually.

BR
Ralf

jow-, 2021-12-11:
Maybe you didn't properly clear your browser cache, it works as expected here now. Disabling and reenabling the list (by changing from disabled to whitelist or disabled to blacklist) will restore all previous items.

derrapf, 2021-12-12:
The video shows exactly what I'm expecting.
Maybe I changed the wrong files?
What I did:
* I replaced  /www/luci-static/resources/form.js  with the on from GIT   [https://github.com/openwrt/luci/blob/f5fbecf132b6c18141896b093935fb4c035ab33d/modules/luci-base/htdocs/luci-static/resources/form.js](here)
* There is another form.js in /rom/www/luci-static/resources which seems to be identical but I cannot change it.
* So currently only  /www/luci-static/resources/form.js is updated
* Then I searched for wireless.js and found it on four locations
* /www/luci-static/resources/view/status
* /www/luci-static/resources/view/network
* /rom/www/luci-static/resources/view/status
* /rom/www/luci-static/resources/view/network
* /overlay/upper/www/luci-static/resources/view/network
* I used the file from [here](https://github.com/openwrt/luci/blob/c534f0671d9a7d0f12d38ece20fcfb3bc19509b2/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js) and replaced /www/luci-static/resources/view/network/wireless.js 
/overlay/upper/www/luci-static/resources/view/network/wireless seems to be the same. 
*I did not change wireless.js in  /www/luci-static/resources/view/status and  /rom/www/luci-static/resources/view/status
Maybe I made an mistake somewhere...
I am not a specialist, so help for dummies would be appreciated.

BR
Ralf

jow-, 2021-12-12:
You need to force clearing your browser cache, otherwise it will continue using the old cached class files.

derrapf, 2021-12-12:
That means I died everything right?
I just tried this, and now it works:
[image]
Thanks very much.
BR
Ralf

derrapf, 2022-07-19:
I update now to OpenWrt 21.02.3.
Once I view the wireless file there is no more list of MAC addresses in there. Nevertheless LUCI shows me the MAC addresses. Are they in a different file now? Where?
BR Ralf

commit c534f067 (2021-12-09): luci-mod-network: wireless.js: retain maclist contents

Fixes: #5579