openwrt/luci #5580: GalanoGrotesque font issue with non-latin symbols (openwrt2020 theme)

svlobanov, 2021-12-05:
Openwrt2020 theme uses GalanoGrotesque font and it looks nice for latin alphabets. GalanoGrotesque doesn't support non-latin symbols so sans-serif is used for non-latin words.

If alphabet contains non-latin symbols (e.g. Greek and Russian) it looks not so nice especially if latin and non-latin symbols are mixed together in one page.

It is especially noticeable in "LED Name" string ("Имя LED", "Όνομα LED"). Non-latin symbols looks much more bold that latin symbols. 

Is there any way not to use GalanoGrotesque font for translations with non-latin symbols? Is it possible to change .css file if Language is changed?

LED settings in English
[image]

LED settings in Greek
[image]

LED settings in Russian
[image]

jow-, 2021-12-05:
The `<html>` element is rendered with a `lang=".."` attribute to allow adding language specific CSS rules. You could try a change like that:

```diff
diff --git a/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css b/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css
index cf320e4aed..a53636d45e 100644
--- a/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css
+++ b/themes/luci-theme-openwrt-2020/htdocs/luci-static/openwrt2020/cascade.css
@@ -10,6 +10,10 @@
        --base-font-size: 16px;
 }
 
+:root[lang="ru"], :root[lang="el"] {
+       --regular-font: "Helvetica";
+}
+
 @font-face {
        font-family: "GalanoGrotesqueW00-Regular";
        src: url("GalanoGrotesqueW00-Regular.woff2") format("woff2");
```

If it works for you (and you can suggest a suitable font), please provide a PR and I'll merge it.

svlobanov, 2021-12-05:
@jow- 
it works perfectly (first one)

>maybe also just adding specific fallback font families is already enough
GalanoGrotesque is a bit different than Helvetica so this solution is not so perfect as first proposed by you

I'm going to prepare PR. it will be ready in a short time

New screenshots for rus and greek:
[image]
[image]

svlobanov, 2021-12-05:
@jow- PR #5581

commit 13949cb8 (2021-12-09): luci-theme-openwrt-2020: fix font issue when mixing latin and non-latin symbols

The GalanoGrotesque font used by the openwrt2020 theme does not support non-
latin symbols.

If latin and non-latin symbols are used together in one line it looks strange
because for latin symbols GalanoGrotesque is used but for non-latin symbols a
fallback font is used (sans-serif).

This patch changes default font to Helvetica for the "bg", "ru", "uk", "el"
and "he" locales.

Original patch was written by Jo-Philipp Wich

Fixes: #5580

[reword commit message, drop "de" from exception list]
closed 2021-12-09 by jow-