openwrt/luci #5582: luci-app-mwan3

5p0ng3b0b, 2021-12-05:
The 'Members', 'Policies' and 'Rules' tabs of the 'Network/Load Balancing' page facilitate moving items up or down the list via 'up' and 'down' buttons. Would it look nicer to merge these into a single 'reorder' button as used in the 'Network/Hostnames' page?

jow-, 2021-12-09:
This is because in 21.02, luci-app-mwan3 has not been converted to client side JS yet but uses older Lua server side rendering. This issue does not apply to master anymore and will be solved once current luci-app-mwan3 is backported to 21.02.

@feckert - do you plan to update luci-app-mwan3 in openwrt-21.02 ?

feckert, 2021-12-10:
I've already ported it to openwrt-21.02 on my local staging branch, but I'm not planning to send it upstream and backport it to the openwrt-21.02. This is already a major change and I don't want to add regressions for the stable branch?

jow-, 2021-12-10:
Fair enough. @5p0ng3b0b - so this issue is changed in master and will be cahnged for the next major OpenWrt release. For 21.02.x it'll remain as-is.

closed 2021-12-10 by jow-