openwrt/luci #5583: Wireguard UI should generate proper QR codes for each peer.

gljones2001, 2021-12-06:
I have tried to use openwrt with wireguard, and it is still a huge pain. Even with the QR code. That reduces somewhere around half of the complexity, but doesn't make it simple.

If you refer to how wire guard works on unraid, you can see how simple it is. You click create peer, and it generates all the numbers and settings you need, and gives you a QR code with all of it. Scan that with your phone, name it, and it just works. No extra settings. No copying any numbers.

Also, setting up computers is still a massive pain as well. You have to copy a bunch of stuff over no matter what. If there was a way for openwrt to generate the zip file that you can import directly into wireguard, that would be nice. This is secondary to the qr code however, because you can export a zip from your phone once you scan the QR code.

I just thought it would be something to look into. I'm not really that great at coding, or else I would help design this. Maybe @danrl you can look into this? It would honestly open up wireguard and make it much more useful on open-source routers.

Thanks,
G

danrl, 2021-12-07:
I'm not sure if I would want a router to generate my cryptographic keys. That said, I don't object the idea and welcome PRs. I won't have time to look into this myself, though.

systemcrash, 2021-12-08:
I took a stab at paste/import. 

(check whichever commit is latest here) https://github.com/systemcrash/luci/tree/wg_peer_paste

nickberry17, 2022-01-01:
> I took a stab at paste/import.
> 
> (check whichever commit is latest here) https://github.com/systemcrash/luci/tree/wg_peer_paste

Hey @systemcrash that's pretty neat!  During testing I ran into a couple of issues that could probably be smoothed out

1. Nothing happens on paste or after updating text (adding spaces, or returns), it only fills in the fields after leaving focus of that box.   Is this a bug?  Is there a way to make this happen as soon as the paste occurs?  I tested on Google Chrome, Firefox and Safari.  Is there a way to handle this case here as a one off without an `onpaste` handler in the form API? 
2. Does the `paste_peer` field need to be visible after everything has been pasted in?  Can visibility be toggled off once it is no longer needed?  This is related to the third point too
3. I already had a peer saved, but it had this box on top.  Can `paste_peer` be invisible if the public key of the peer is not the empty string? e.g. `o.depends({ public_key: '', "!contains": true })` http://openwrt.github.io/luci/jsapi/LuCI.form.AbstractValue.html#depends
4. Nitpick: the fields are not cleared if the box is cleared

systemcrash, 2022-01-01:
> Hey @systemcrash that's pretty neat! And it is good that it does not seem to fill in the fields with invalid data either (I tried pasting some random stuff in). During testing I ran into a couple of issues that could probably be smoothed out
> 

Cool! Thanks for the input. 

>     1. Nothing happens on paste or after updating text (adding spaces, more text, or returns), it only fills in the fields after leaving focus of that box.   Is this a bug?  Is there a way to make this happen as soon as the paste occurs?  I tested on Google Chrome, Firefox and Safari.  @jow-  Is there a way to handle this case here as a one off without an `onpaste` handler in the form API?

Expected. One way: https://openwrt.github.io/luci/jsapi/LuCI.ui.Textarea.html#isChanged 

>     2. Does the `paste_peer` field need to be visible after everything has been pasted in?  Can visibility be toggled off once it is no longer needed?  This is related to the third point too
 
Not necessary. But trying to make the field's behaviour deterministic will likely confuse users and frustrate them rather than always just being available. 

>     3. I already had a peer saved, but it had this box on top.  Can `paste_peer` be invisible if there is a valid key already in the field e.g. `o.depends({ public_key: /^[A-Za-z0-9+/]{42}[A|E|I|M|Q|U|Y|c|g|k|o|s|w|4|8|0]=$/, "!reverse": true });`
>        regex looking for a WireGuard public key (source: https://lists.zx2c4.com/pipermail/wireguard/2020-December/006222.html) Could also be used as a custom datatype validator for WireGuard keys instead of the more generic validateBase64?

Perhaps you already have a peer, like a VPN provider, and you need to paste in a fresh random key once in a while. You then have your work cut out for you to make the field visible to paste. So any field or mechanism as part of this maintenance should, I think, always be easily reachable/available. Although, granted, see next point. Wireguard keys are base64 encoded - so validating for anything else is pointless. And unless you know both ends, it is best to leave it up to `wg` to determine whether they keys are legit. Who knows (this would never happen but the principle stands) - maybe base64 standard is refreshed and/or the encoding scheme changed. 

>     4. Nitpick: the fields are not cleared if the box is cleared

Intentional - once pasted, its job is done. Using the field's lifecycle to manage everything else's behaviour is cumbersome. Though perhaps it's better to just scrap that entry and paste into a new endpoint...

@jow- rewrote the endpoint manager to be a GridSection or something anyway, so it is just a click away.

jow-, 2022-07-22:
Fixed by 9ba20645b01aacb391ab8c69f57b3da60db8f630

closed 2022-07-22 by jow-

gljones2001, 2022-07-22:
NO WAY.

They actually implemented the better Wireguard system. I cant believe MY suggestion actually made its way i to the OFFICIAL codebase.

AWESOME 👍

Thanks so much btw. This makes Wireguard SOOO much more useful than the previous setup.