openwrt/luci #5585: Broadcom: OpenWRT userland running on DD-WRT kernel and drivers

Lanchon, 2021-12-06:
somewhat off-topic, sorry for that, but i didn't know where else to post.

it is a sad reality that Broadcom will never free its current and past silicon. if market forces ever pushes Broadcom to opensource, they will only open the then current devices. it is also very sad that so many Broadcom routers out there are then stuck with DD-WRT. DD-WRT sort of works; but code quality is lacking, docs are terrible, contributing is hard, and so it has many longstanding issues.

i was wondering if there are any efforts ongoing to run OpenWRT userland on top of DD-WRT, or whether such a thing could be organized. although clearly the idea does not fit with core OpenWRT values, it's usefulness for many (eg, those running stock or DD-WRT) is undeniable.

i am mainly thinking of stock DD-WRT configured with wireless and most things disabled and a startup script that kills everything and switches to OpenWRT userland in a R/W partition (JFF2 or USB). closed-source userland parts (eg, `wl`) would be ported over to the new userland at switching time. this sidesteps the issue of distributing of non-free components, as users would get stock DD-WRT for first install or to upgrade the kernel. OpenWRT would keep all settings in files and leave the nvram area for exclusive use by stock DD-WRT during boot.

any and all comments welcome. thanks!

jow-, 2021-12-06:
As far as I am aware, there are no such efforts. The biggest hurdle will be emulating the cfg80211 API on top of wl.ko since we're about to kill the proprietary wl.ko backend soon.

Lanchon, 2021-12-06:
oh no! i only thought of this because you guys support wl. do you think a PoC might alter this decision?

jow-, 2021-12-07:
A proof of concept will not lead to OpenWrt developers suddenly wanting to spend effort working with the deficiencies of closed drivers, old kernels and proprietary APIs.

If someone can contribute a high quality integration and is committed to maintain it in the long run then we could maybe merge it, but I don't see the already quite small contributor base to suddenly start working on a hardware base that is not able to run unmodified OpenWrt.

Off the top of my head, the things you'd need to solve are:
 - Wireless configuration 
   - LuCI (still) uses an intermediate abstraction library (libiwinfo) which can work with multiple driver backends. Someone needs to work on a high quality, fully featured [Broadcom driver backend](https://lxr.openwrt.org/source/iwinfo/iwinfo_wl.c) that covers all important use cases - the existing one is limited and bugged. This likely requires NDA level access to Broadcom driver internals or at least a little bit of experience in order to extract the relevant data out of proprietary ioctl() interfaces
   - OpenWrt integration, the broadcom-wl [wireless configuration integration for netifd](https://github.com/openwrt/openwrt/blob/master/package/kernel/broadcom-wl/files/lib/wifi/broadcom.sh) needs to be revamped and extended to properly cover all use cases (multi-SSID, MAC filter etc.)
 - Network configuration
   - swconfig or DSA support is required kernel side for VLAN setup, note that swconfig is on its way out in OpenWrt though, so in the long run LuCI support for it will likely go stale as well

Lanchon, 2021-12-07:
hey thanks for all the info!

i was looking closer at stuff. turns out DD-WRT is running linux 4.4.293 which is a longterm from 2021-11-26 that will EOL by Feb, 2022. im assuming OpenWRT will not tolerate a kernel downgrade, so this would force me to use OpenWRT 17.01.7 from June 2019, which is long unsupported.

maybe DD-WRT transitions to a newer longer next Feb, and it makes more sense to review this then. right now it does not seem very useful.

jow-, 2021-12-09:
Closing this as it likely won't manifest without external development efforts. You could try to gain more momentum in the DD-Wrt and/or OpenWrt forums.

closed 2021-12-09 by jow-