openwrt/luci #5593: wireless settings ipq40xx fritzbox 7530 snapshot r18302

francisuk1989, 2021-12-11:
Today i installed the latest snapshot build of OpenWrt SNAPSHOT r**18302-ea49690ff4 / LuCI Master git-21.343.55550-008bd89** Dated **Fri Dec 10**

When installing Luci via SSH, I see no theirs no Wireless configuration but everything eles seem ok, However SSHing

`root@fritzbox:~# wifi up`

Works from a 21.02.1 file inside /etc/config/ not sure if there was already a wireless file in there but not sure.

jow-, 2021-12-11:
Not a LuCI problem, please report at bugs.openwrt.org

closed 2021-12-11 by jow-

francisuk1989, 2021-12-13:
@jow- 
Installed today's snapshot (13 December 2021) and the Wireless tab is now showing. however i get one or two errors when installing Luci

Configuring liblucihttp0.
Configuring cgi-io.
Configuring liblua5.1.5.
Configuring lua.
Configuring luci-lib-nixio.
Configuring luci-lib-ip.
Configuring luci-lib-jsonc.
Configuring liblucihttp-lua.
Configuring luci-lib-base.
Configuring luci-app-opkg.
Configuring rpcd.
`uci: Entry not found`
Configuring libubus-lua.
Configuring libiwinfo-lua.
Configuring rpcd-mod-file.
Configuring rpcd-mod-luci.
Configuring luci-base.
Configuring luci-mod-system.
Configuring luci-theme-bootstrap.
/luci-static/bootstrap
/luci-static/bootstrap-dark
/luci-static/bootstrap-light
`uci: Entry not found`
Configuring luci-mod-status.
Configuring luci-app-firewall.
Configuring luci-proto-ppp.
Configuring rpcd-mod-iwinfo.
Configuring luci-mod-network.
`uci: Entry not found`
Configuring luci-mod-admin-full.
Configuring luci-proto-ipv6.
Configuring uhttpd.
4+0 records in
4+0 records out
Configuring uhttpd-mod-ubus.
uci: Entry not found
Configuring rpcd-mod-rrdns.
Configuring luci.