openwrt/luci #5595: QoS over Nftables : Realtime Rate

NotANewNick, 2021-12-12:
Hello,
I have experienced the Realtime Rate of QoS over Nftables not actually showing the Rate and instead accumulating change over time.

[image]
[image]

nft-qos config:
`
config default 'default'
        option limit_mac_enable '0'
        option priority_enable '0'
        option limit_enable '0'
`

Is anyone able to suggest a fix or perhaps point me to where I might be able to fix it my self?

wulfy23, 2021-12-14:
https://github.com/openwrt/luci/blob/d5385fa93b5d802f780e1e2393cc20dd633cdcd9/applications/luci-app-nft-qos/luasrc/view/nft-qos/rate.htm#L32

NotANewNick, 2021-12-14:
That seems like an unlikely culprit. From the table I can see multiple values for the same IP which to is telling me that something is not populating the table correctly.

The code you posted seems to be the interface part and not the data processing part.

wulfy23, 2021-12-16:
yeah... there are duplicate values... that's related to the core package not luci

easy fix pr's for the packages repo are another story... there's a thread on the forum if you'd like a fix for that to check if it helps with your values...

NotANewNick, 2021-12-16:
Building now. I'll close the ticket in case this works.
Thank you!

NotANewNick, 2021-12-17:
So far so good. Thank you.

closed 2021-12-17 by NotANewNick

wonson, 2023-07-27:
Thanks @wulfy23 
Your fix is good, but do you know which repo should we report the issue and get it into official release?

Now we need to apply the patch after every sysupgrade :(