openwrt/luci #5604: Unable to access page, error reported

money-outlook, 2021-12-15:
/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:59: attempt to index local 'value' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:59: in function 'getProviderName'
	/usr/lib/lua/luci/model/cbi/https-dns-proxy.lua:92: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1355: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1026: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

stangri, 2021-12-15:
Hello, could you please post the output of:

```sh
cat /etc/banner
opkg list-installed | grep https-dns-proxy
cat /etc/config/https-dns-proxy
```

money-outlook, 2021-12-16:
_______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.1, r16325-88151b8303
 -----------------------------------------------------

https-dns-proxy - 2021-09-27-1
luci-app-https-dns-proxy - 2021-07-29-1

config main 'config'
        option update_dnsmasq_config '*'
        option force_dns '1'
        list force_dns_port '53'
        list force_dns_port '853'
# ports listed below are used by some
# of the dnscrypt-proxy v1 resolvers
#       list force_dns_port '553'
#       list force_dns_port '1443'
#       list force_dns_port '4343'
#       list force_dns_port '4434'
#       list force_dns_port '5443'
#       list force_dns_port '8443'

config https-dns-proxy
        option bootstrap_dns '1.1.1.1,1.0.0.1'
        option resolver_url 'https://cloudflare-dns.com/dns-query'
        option listen_addr '127.0.0.1'
        option listen_port '5054'
        option user 'nobody'
        option group 'nogroup'

config https-dns-proxy
        option bootstrap_dns '8.8.8.8,8.8.4.4'
        option resolver_url 'https://dns.google/dns-query'
        option listen_addr '127.0.0.1'
        option listen_port '5053'
        option user 'nobody'
        option group 'nogroup'

money-outlook, 2021-12-17:
I think there is a problem with one of the DOH configuration files, such as "com.controld.freedns.family.lua".

switch the configuration file a few more times on the error, but if you remove these configuration files, leaving only google, cloudflare, there will be no problem

stangri, 2021-12-20:
I can't reproduce the issue with your exact config/software versions.

closed 2021-12-24 by stangri