openwrt/luci #5610: Enabling HTTPS via Luci-SSL package seems broken in latest master snapshot

PalebloodSky, 2021-12-18:
Performed a clean install of OpenWrt (WRT32X router) Master snapshot dated 12/18/2021 OpenWrt SNAPSHOT r18371-5a4685cfa2:

SSH in: `opkg update && opkg install luci-ssl`

Attemping to enable via LuCI in System > Administration > HTTP(S) Access. Clicked enable, then Save and Apply. It just keeps loading, does not commit the change.

Enabling HTTPS via SSH and UCI worked as expected:
```
uci set uhttpd.main.redirect_https=1
uci commit uhttpd
service uhttpd reload
```

So it appears that the LuCI package is broken.  I double checked the guide on the wiki and it seems consistent with what I attemped to do above, I would have updated the guide if it appeared out of date: https://openwrt.org/docs/guide-user/luci/luci.secure

Thanks.

SiNONiMiTY, 2021-12-23:
It seems there is a bug on the page reload mechanism. During save, manually going back to the page will save the configuration. Nevertheless, this is a bug.

translucentfocus, 2021-12-27:
Also, if compiling firmware from scratch, with LuCI built-in and a first run script added to `package/base-files/files/etc/uci-defaults` 
with the following contents
```
# Force https:// for WebGUI to ensure security when accessing on Wi-fi
set uhttpd.main.redirect_https='on'
commit uhttpd

```
then the issue does not appear.

You may want to try the syntax above and see if that works intead.

SiNONiMiTY, 2021-12-27:
I just checked the Browser DEV tools and saw that there are errors regarding SSL
[image]

So, the page reload mechanism is working fine, but the page cannot be loaded again since there is an error on the certificate.

jow-, 2022-07-12:
CLosing this as I assume it is/was an issuwer with the underlying SSL support libraries in OpenWrt.

closed 2022-07-12 by jow-