openwrt/luci #5625: Add support for OpenVPN's topology option

jerry-yuan, 2022-01-02:
I have tried to add an additional option named topology to config file like these:
[image]
It work well after openvpn instance restart even there is no any items displayed in LuCI.

But if I tried to modify any options via LuCI, these additional options will disappear.

The topology option is used to define the tunnel device's mode and will affect the vpn clients to get hostname resolution from dnsmasq service which only support relove hostnames for the request from local subnet by default.

I want to know if there's possible to support those options which has not mapped to LuCI displayed items or make it easy to add some custom options (like advanced configuration edit mode).

mrendo, 2022-01-02:
Hello,

In advance go to Networking add dev-type and make sure it says tun it should do by default then go back to the drop down menu and topology will be on the bottom of the menu. Hope that helps.

Mr E

jerry-yuan, 2022-01-03:
> Hello,
> 
> In advance go to Networking add dev-type and make sure it says tun it should do by default then go back to the drop down menu and topology will be on the bottom of the menu. Hope that helps.
> 
> Mr E

Thanks for your reply and with following your suggestions I have successfully added the `topology` option.

By the way, I'd like to know if there is any documents to reference which has explained the dependencies between the options?
(like if option `b` need to be set after `a` has been set to some value)

closed 2022-01-08 by jerry-yuan

Torbins, 2023-11-25:
Wow, that's super counterintuitive.