openwrt/luci #5629: OpenWrt 21.02.1: Failed to change the system password

alex9434, 2022-01-04:
I get the error "Failed to change the system password" on Luci on OpenWrt 21.02.1 on arm64 (Raspberry Pi 4). I use a system that I have compiled using buildroot and that uses NGINX for Luci. Any idea what can cause this
[image]
?

jow-, 2022-01-04:
Issue the following command on the cli and provide the output here:
```
ubus call luci setPassword '{ "username": "root", "password": "test" }'
```

alex9434, 2022-01-15:
I just get the following message: 
```
{
        "result": false
}

```

alex9434, 2022-01-16:
Connected to this issue is another issue: If I change the root password using passwd on the command line, I can no longer log into LuCI. It will always tell me the password is incorrect. The link between LuCI and the system password seems to be broken.

jow-, 2022-01-25:
The shadow-passwd applet (`/usr/bin/passwd`) will store blowfish (`$2a$...`) ciphers in `/etc/shadow` which musl libc's `crypt()` implementation does not support by default, hence the failure to log into LuCI. This is nothing that can be fixed on the LuCI side.

I pushed another fix though which will ensure to invoke Busybox `passwd` when changing the system password via LuCI, in this case it is guaranteed that a supported password hash is written to `/etc/shadow`.

closed 2022-01-25 by jow-

hmh, 2022-01-30:
This isn't a bug in LuCI, but shouldn't the *defaults* in shadow-passwd and musl be kept in sync?  Without doing an in-depth look at the issue, it seems like there's a bug in shadow-passwd packaging if it has defaults that are incompatible with the defaults for the default libc (musl)...

jow-, 2022-01-30:
Yes, they should be kept in sync, but this outside of the scope of LuCI and this repo.

commit 22e2bfb0 (2022-01-25): luci-base: explicitly invoke busybox applet for password change

Ensure to invoke the Busybox `passwd` applet to change the system password
in a non-interactive manner. Non-Busybox variants may not take the new
password input from stdin or use password hashes which are not supported
by musl's `crypt()` implementation by default.

Fixes: #5629
commit 4ec18ec7 (2022-01-25): luci-base: explicitly invoke busybox applet for password change

Ensure to invoke the Busybox `passwd` applet to change the system password
in a non-interactive manner. Non-Busybox variants may not take the new
password input from stdin or use password hashes which are not supported
by musl's `crypt()` implementation by default.

Fixes: #5629

(cherry picked from commit 22e2bfb09667e167a93b5eca4487ac41669126d9)