openwrt/luci #5632: Unable to select correct 5GHz channel numbers for 40MHz and 80MHz widths

jasonemann, 2022-01-06:
Model | Ubiquiti UniFi AC Pro
Architecture | Qualcomm Atheros QCA956X ver 1 rev 0
Firmware Version | OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.295.67054-13df80d
Kernel Version | 5.4.154

When configuring a 5GHz wireless network, the dropdown list of channels does not list the correct channels when 40MHz and 80MHz widths are selected.

Regardless of which width is selected, only the 20MHz channel numbers are listed: 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149, 153, 157, 161, 165, 169, 173.

When 40MHz width is selected, the channel list should be: 34, 38, 46, 54, 62, 102, 110, 118, 126, 134, 142, 151, 159, 167, 175.

When 80MHz width is selected, the channel list should be: 42, 58, 106, 122, 138, 155, 171.

[image]
[image]

jow-, 2022-01-06:
That is not how the OpenWrt/hostapd backend works. It expects a 20Mhz primary channel number and will select the center frequency automatically. The list exposed by LuCI is taken from nl80211 (see `iw list`). The channel configuration *does not* specify center frequencies.

Chosing e.g. channel 36 and an 80MHz width will result in `vht_oper_centr_freq_seg0_idx=42` (center channel number 42 = 5210 MHz). So you can think of the channel number as the "lower edge" of effectively used 40/80/160MHz band.

closed 2022-01-06 by jow-

jasonemann, 2022-01-06:
Thanks for the clarification.

jow-, 2022-01-06:
If you're curious, you can find the calculations here:
https://github.com/openwrt/openwrt/blob/master/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh#L233