openwrt/luci #5638: staging_dir/host/bin/sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory

zhangguanzhang, 2022-01-08:
Error when install with image buider:
```
$ make image PACKAGES="$PACKAGES ${extra_pkgs}" FILES="files"
...
...
Configuring luci-i18n-upnp-zh-cn.
Configuring kmod-usb-net-sr9700.
Configuring coreutils-cat.
Configuring kmod-dummy.
Configuring kmod-fs-squashfs.
Configuring kmod-net-prism54.
Configuring luci-app-docker.
Configuring iperf3.
Configuring tmate.
Configuring kmod-p54-usb.
Configuring luci-i18n-firewall-zh-cn.
Configuring luci-i18n-docker-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Configuring libnl-genl200.
Configuring ipvsadm.
Configuring luci-proto-ncm.
Configuring kmod-net-rtl8192su.

Finalizing root filesystem...
/root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
/root/op/openwrt-imagebuilder-rockchip-armv8.Linux-x86_64/staging_dir/host/bin/sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
Enabling AdGuardHome
Enabling aliyundrive-webdav
Enabling aria2
Enabling arpbind
Enabling autoreboot
Enabling boot
Enabling cgroupfs-mount
...
...
```

zhangguanzhang, 2022-01-08:
```
root@guan:/home/guanzhang/lede# find -type f -exec grep -Pzil 'sed.+?/usr/lib/lua/luci/dispatcher.lua' {} \;
./package/lean/luci-app-filetransfer/root/etc/uci-defaults/luci-filetransfer
./feeds/others/luci-app-argonne-config/root/etc/uci-defaults/luci-argonne-config
./feeds/others/luci-theme-argonne/Makefile
./feeds/others/luci-theme-argonne/root/etc/uci-defaults/90_luci-theme-argonne
./feeds/others/luci-theme-edge/Makefile
./feeds/others/luci-theme-edge/root/etc/uci-defaults/30_luci-theme-edge
```
似乎是某些包里的依赖没有依赖 luci-base 造成提前执行sed命令修改  luci-base 的文件

hnyman, 2022-01-08:
Looks like you have lots of strange unsupported packages there  (luci-theme-argonne, luci-theme-edge, luci-app-filetransfer, luci-app-argonne-config)  Probably some of those is not in sync with the up-to-date LuCI

zhangguanzhang, 2022-01-09:
I'm making some changes to these Makefiles, I'll see if it works later

zhangguanzhang, 2022-01-09:
I found this section `postinst`  while be `etc/uci-defaults/90_luci-theme-argonne`:
```
define Package/luci-theme-argonne/postinst
#!/bin/sh
sed -i ":a;$!N;s/tmpl.render.*sysauth_template.*return/local scope = { duser = default_user, fuser = user }\nlocal ok, res = luci.util.copcall\(luci.template.render_string, [[<% include\(\"themes\/\" .. theme .. \"\/sysauth\"\) %>]], scope\)\nif ok then\nreturn res\nend\nreturn luci.template.render\(\"sysauth\", scope\)/;ba" /usr/lib/lua/luci/dispatcher.lua
sed -i ":a;$!N;s/t.render.*sysauth_template.*return/local scope = { duser = h, fuser = a }\nlocal ok, res = luci.util.copcall\(luci.template.render_string, [[<% include\(\"themes\/\" .. theme .. \"\/sysauth\"\) %>]], scope\)\nif ok then\nreturn res\nend\nreturn luci.template.render\(\"sysauth\", scope\)/;ba" /usr/lib/lua/luci/dispatcher.lua
[ -f /usr/lib/lua/luci/view/themes/argon/out_header_login.htm ] && mv -f /usr/lib/lua/luci/view/themes/argon/out_header_login.htm /usr/lib/lua/luci/view/header_login.htm
rm -Rf /var/luci-modulecache
rm -Rf /var/luci-indexcache
exit 0
endef
```

so I hack this section:
```
if [ -f ./feeds/others/luci-theme-argonne/Makefile ];then
    SED_NUM=$( grep -Pn '^\s*define\s+Package/\S+/postinst' ./feeds/others/luci-theme-argonne/Makefile |  awk -F: '$0~":"{print $1}')
    if [ -n "SED_NUM" ];then
        sed -i "$[SED_NUM+2]i [ ! -f /etc/openwrt_release ] && exit 0" ./feeds/others/luci-theme-argonne/Makefile
    fi
fi
```

closed 2022-01-09 by zhangguanzhang