openwrt/luci #5644: [21.02.1] frontend hangs on firmware flash

jjm2473, 2022-01-12:
macOS 10.15.7, Chrome  96.0.4664.110

I connect openwrt from WAN port,
after flashing, luci will wait for `luci-static/resources/icons/loading.gif` with 1 second timeout, but luci fill browser network request queue, make all request failed after 15 seconds, so luci never seen router online.

feckert, 2022-01-12:
Is it possible that you got a different IP? The wan port is configured to dhcp. Is your firewall correct and do you have the port open?

jow-, 2022-01-12:
Also HTTPS issues (new cert, downgrade from HTTPS to HTTP etxc.) might prevent loading the sentinel file successfully. All in all this is a minor convenience feature that is supposed to automatically reload the page if the router is found to be alive.

I don't see any good fix to it, so I'll close this issue.

closed 2022-01-12 by jow-

jjm2473, 2022-01-13:
Maybe we should use ajax to check router online, we can set timeout on ajax request. We using a img tag to load sentinel file 
 in current implemention, this may fill the request queue.

jow-, 2022-01-13:
Ajax is subject to various cross domain restrictions and other headaches.

jjm2473, 2022-01-13:
oh, i see