openwrt/luci #5646: Error creating interfaces

castillofrancodamian, 2022-01-12:
I have Snapshots and have noticed that when creating an interface, it is not saved when pressing the "Create interface" button. It also does not save the entered data such as IP or any settings. It only lets me exit if I press the "Back" button and go back to the interface to just configure it there.

[image]
[image]
[image]
[image]
[image]
[image]

jow-, 2022-01-12:
Check the browser debug console for errors.

castillofrancodamian, 2022-01-13:
[image]
[image]
[image]

closed 2022-01-19 by jow-

castillofrancodamian, 2022-03-27:
I have noticed the same error in the DDNS app when creating a new service.

commit 35df2ada (2022-01-19): luci-mod-network: fix creating new interfaces

Fixes: #5646
Fixes: 1c798d1a08 ("luci-base: form.js: allow nesting Grid and Table sections")