openwrt/luci #5661: Insufficient permissions to read UCI configuration. OpenVPN luci

openfnord, 2022-01-25:
# Issue template:
1. Please make sure that the issue subject starts with `<package-name>:`
2. Issues related to end of life (EOL) releases are not supported or maintained and will be closed
3. Remove lines from top till here and fill in the following questions
***

## Steps to reproduce:
Have an account with proton vpn and download their template for openvpn
Use luci openVPN web gui and upload the config file. Edit the file in the top box, and add the file name for
the credentials file that is pasted into the bottom box.
Save it.
Go to the luci-open-vpn interface and one sees a message in yellow : 
"Insufficient permissions to read UCI configuration"

## Actual behavior:
"Insufficient permissions to read UCI configuration"

## Expected behavior:
no message of 
"Insufficient permissions to read UCI configuration"
Storage of settings and proper vpn operation

## Additional Information:
A resolution has been proposed which I tried on my Netgear r6220. 
Here the warning about insufficient rights disappeared. The VPN needs to be tested,
but the message disappeared.

See here and take the fix from there:
https://github.com/openwrt/luci/issues/4370
https://web.archive.org/web/20211116014744/https://forum.openwrt.org/t/insufficient-permissions-to-read-uci-configuration/106124

the modified map.html proposed above (archive.org) helped to resolve the issue.
Please integrate it into the distribution filles

"
root@Peter:/usr/lib/lua/luci/view/cbi# cat map.htm 
<%- if firstmap and messages then local msg; for _, msg in ipairs(messages) do -%>
        <div class="alert-message warning"><%=pcdata(msg)%></div>
<%- end end -%>

<% if firstmap and readable then %>
<div class="cbi-map" id="cbi-<%=self.config%>"<%=ifattr(not writable, "style", "opacity:.6; pointer-events:none")%>>
        <% if self.title and #self.title > 0 then %>
                <h2 name="content"><%=self.title%></h2>
        <% end %>
        <% if self.description and #self.description > 0 then %>
                <div class="cbi-map-descr"><%=self.description%></div>
        <% end %>
        <% if self.tabbed then %>
                <div>
                        <% for i, section in ipairs(self.children) do
                               tab = section.section or section.sectiontype %>
                                <div class="cbi-tabcontainer"<%=
                                        attr("id", "container.m-%s.%s" %{ self.config, tab }) ..
                                        attr("data-tab", tab) ..
                                        attr("data-tab-title", section.title or tab)
                                %>>
                                        <% section:render() %>
                                </div>
                        <% end %>
                </div>

                <% if not self.save then -%>
                        <div class="cbi-section-error">
                                <% for _, section in ipairs(self.children) do %>
                                        <% if section.error and section.error[section.section] then -%>
                                                <ul><li>
                                                        <%:One or more invalid/required values on tab%>:&nbsp;<%=section.title or section.section or section.sectiontype%>
                                                </li></ul>
                                        <%- end %>
                                <% end %>
                        </div>
                <%- end %>
        <% else %>
                <%- self:render_children() %>
        <% end %>
</div>
<% elseif firstmap then %>
<div class="alert-message warning"><%:Insufficient permissions to read UCI configuration.%></div>
<% end %>
"

root@Peter:/usr/lib/lua/luci/view/cbi# diff _map_orig.htm map.htm 
5c5
< <% if readable then %>
---
> <% if firstmap and readable then %>
42c42
< <% else %>
---
> <% elseif firstmap then %>
root@Peter:/usr/lib/lua/luci/view/cbi# 

So my Netgear r6220, running OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-21.295.67054-13df80d
does not show the error after using the modified map.htm which was proposed before.
So please add the modification to the distribution code this time and do not close the issue.

jow-, 2022-01-25:
Duplicate of #4370

closed 2022-01-25 by jow-

openfnord, 2022-01-25:
The problem has to be solved, closing does not help as people would need to resort to untar some fancy stuff into their openwrt as openwrt does not take the good changes

See /usr in the tarball of my openwrt_stuff
https://github.com/openfnord/openwrt_foo

jow-, 2022-01-26:
The problem has been solved. See ticket #4370 which this ticket is a duplicate of.