openwrt/luci #5664: HTTP:// can't login ,  HTTPS get browser warning about being unsafe

peterwillcn, 2022-01-26:
OpenWRT 21.02.0 Stable

Steps to reproduce:

Install OpenWRT 21.02.0 Stable as a fresh install.
Visit 192.168.1.1 and sign into Luci (with no password since it is a fresh install), go to System → Administration and assign a password
Close the browser window
Login through SSH and enable HTTPS-redirect for Luci (as mentioned in the release notes: https://openwrt.org/releases/21.02/notes-21.02.0)
Open a new browser window in Incognito-mode
Visit 192.168.1.1 (HTTPS get browser warning about being unsafe)
Try to sign http:// into Luci (should get message that the user/password is incorrect)
I have tried clearing my browser cookies, tried disabling the HTTPS redirect, restarted the router manually (power switch on device), tried using the “passwd” command to reset the password (and then restarted the router manually).

No matter what, I cannot login into the router ever again after enabling HTTPS redirect for Luci (even if the redirect is turned off and router restarted). I can, strangely, log into the router through SSH with the password I set for root.

I had to login through SSH and run the (cant remember exact) command to reset the router as if I just installed OpenWRT in order to gain access again.

jow-, 2022-01-26:
This is a common browser quirk. As long as you still have a `sysauth` cookie pegged to a (former) HTTPS session, logging in via HTTP is impossible since the HTTP provided sysauth cookie is not allowed to overwrite the stale HTTPS one.

peterwillcn, 2022-01-26:
@jow- clear cook fix, Thanks

closed 2022-01-26 by peterwillcn