openwrt/luci #5673: luci-mod-status: firewall status can't show anything

time-river, 2022-02-04:
In luci Status Firewall page, nothing shows in snapshot version.

I read the configure and the [src](https://github.com/openwrt/luci/blob/master/modules/luci-mod-status/htdocs/luci-static/resources/view/status/iptables.js), find OpenWrt is migrating nftables from iptables, `firewall4` has been default selection instead of `firewall`. However, nftables use lower case, such as `input`, `forward`, and `output` instead of `INPUT`, `FORWARD`, and `OUTPUT` used by `iptables`. Therefore, nothing will be lookup. And `iptables` isn't installed default.

Switch to nftables from iptables, please.

hnyman, 2022-02-04:
There is work-in-progress for a nfables status page.
See PR #5652

jow-, 2022-02-15:
An nftables status page has been implemented with 4fc3cd8e42ff9245f028c521307fa45e6a0c76af

closed 2022-02-15 by jow-