openwrt/luci #5678: [BUG] nftables ports opened by miniupnpd not shown on status page or upnp luci app page

edrikk, 2022-02-08:
Master build as of [ae75541](https://github.com/openwrt/openwrt/commit/ae75541594c1802be16140aeb59254e1c75fe7a8) shows that the miniupnpd ports are correctly opened via fw4 rules.

However, both the Luci status -> overview page’s “Active UPnP Redirects” as well as luci-app-upnp package’s “Active UPnP Redirects” sections are empty.

Expected behaviour is of course for the redirects to appear on these pages when nftables are used for miniupnpd.

urbenlegend, 2022-02-15:
I am also seeing this issue on my end. In addition to this, it seems that sometimes port mappings fail to get removed properly, with errors shown in the logs saying as much.

jckefan, 2022-07-21:
The issue still persists in July.

firewire10000, 2022-08-08:
I've just compiled 22.03.0-rc6 with miniupnpd 2.3.0 (modified the Makefile to download the latest version) and I too am not seeing anything show under 'Active UPnP Redirects' table of LuCI nor am I getting any open ports. I have even applied the patch mentioned in the forum post [here](https://github.com/openwrt/packages/issues/17871#issuecomment-1207162628) whilst compiling the firmware and no difference.

elliotclee, 2022-09-04:
I came here for what I think is this bug... In /usr/libexec/rpcd/luci.upnp (part of the luci-app-upnp package), the get_status method seems to try running iptables, which of course doesn't work with firewall4/nftables. I don't know enough to submit a patch but it seems like this could be part of the problem. I am running 22.03-rc6.

elliotclee, 2022-09-04:
I just checked the master and openwrt-22.03 branches of https://github.com/openwrt/luci/blob/master/applications/luci-app-upnp/root/usr/libexec/rpcd/luci.upnp and it is still broken there. @Ansuel wrote the original code.

jow-, 2022-09-08:
Fixed by #5961

closed 2022-09-08 by jow-