openwrt/luci #5683: [BUG] Configuring both 5GHz Client and 5GHz Access Point will cause 5GHz Wi-Fi "disappear"

ngjiaxue, 2022-02-12:
If there is no 5GHz Client configure (Only 5GHz Access Point & 2.4GHz Access Point)
[image]
[image]

If configuring both 5GHz Client and 5GHz Access Point will cause 5GHz Wi-Fi "disappear"
[image]
[image]

jow-, 2022-02-12:
This does not look like a LuCI issue, the networks appear to be configured correctly as far as LuCI settings are concerned. Maybe a low level driver bug or your phone not properly picking up the beacons of the 5GHz radio or not supporting channel 157 which is restricted in some regions.

Closing this as this cannot be solved in LuCI

closed 2022-02-12 by jow-

ngjiaxue, 2022-02-13:
Hi,

So, may I know who should I contact regarding to this issue? I really need
help...

Thank you.

Best Regards,
Ng Jia Xue

hnyman, 2022-02-13:
It might be better to seek help in the Openwrt discussion forum.   https://forum.openwrt.org/
(and you should also provide details from the system log, that which king of error messages are shown)

OpenWrt wifi bugs should currently be reported at the OpenWrt bug tracker:  https://github.com/openwrt/openwrt/issues

There are also similar bugs for the QCA988x wifi chip mentioned in the old bug tracker, so you are not the first to notice the problem with that wifi chip:
https://bugs.openwrt.org/index.php?do=details&task_id=1158
https://bugs.openwrt.org/index.php?do=details&task_id=3569