openwrt/luci #5685: luci-app-firewall: error in destination zone selection in port forwarding

castillofrancodamian, 2022-02-15:
## Steps to reproduce:

1. go to: Network → Firewall → Port Forwards → Add
2. create a new rule with "wan" for "Source zone" and "unspecified" for "Destination zone". Add any port to test
3. Save & Apply
4. press the "Edit" button to verify that the zone in "Destination zone" changed to "lan".

## Expected behavior:

Let the change be applied.

## Additional Information:

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r18801-56256259a1'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r18801-56256259a1'
DISTRIB_TAINTS=''

castillofrancodamian, 2022-02-15:
[image]
[image]
[image]
[image]
[image]

jow-, 2022-02-15:
Okay, so it`s _unspecified_ which does not work. I tried selecting a specific zone and that was fine.

commit 21784446 (2022-02-15): luci-app-firewall: fix setting unspecified zone on forwardings

Fixes: #5685
closed 2022-02-15 by jow-