openwrt/luci #5689: luci-app-statistics: graphs using BootstrapDark nearly impossible to view

rhuang2014, 2022-02-18:
Recent update to luci, the statistics graphs are near impossible to see when using BootstrapDark them.

New Look:
[image]

Previous Look:
[image]

## Steps to reproduce:
1. go to: System → System → Language and Style → Design → BootstrapDark
2. Save & Apply
3. go to: Statistics -> Graphs -> Processor ( or any other graph, i.e Ping )

## Expected behavior:
Maybe have a slightly lighter background color for the graphs when Dark mode is selected?

## Additional Information:
OpenWrt 21.02-SNAPSHOT r16497-d1c15c41d9
LuCI openwrt-21.02 branch (git-22.046.85957-59c3392)

rhuang2014, 2022-02-18:
Hi @jow- , Do you know which commit or file(s) I can revert to get the old behavior (image with white background) back?  I try looking over the commit history, but I was not able to identify the commit I can revert to get the old view back.  I appreciate your response, I guess each person has their own opinion on what's acceptable and what's not to them.

rhuang2014, 2022-02-19:
The alternative solution didn't seem to work for me.  It seems the custom_style.css has no effect.  Here are the changes based on my understanding.
```css
# cat /www/luci-static/resources/custom_style.css
[data-darkmode="true"] [data-page="admin-statistics-graphs"] [data-plugin] img {
    filter: none !important;
}
# cat /www/luci-static/resources/preload/custom_css_loader.js
'use strict';
'require baseclass';

return baseclass.extend({
    __init__: function() {
        document.querySelector('head').appendChild(E('link', {
            rel: 'stylesheet',
            href: L.resource('custom_style.css')
        }));
    }
});
```

Instead of dropping the CSS rule, I set the value for `invert` and `hue-rotate` to `0` like below seems to restore to previous look for me.
```css
 [data-darkmode="true"] [data-page="admin-statistics-graphs"] [data-plugin] img { 
 	filter: invert(0%) hue-rotate(0deg); 
 }
```

How difficult or is it possible to add an extra checkbox to the `Language and Style` tab if checked, it will set `0%` for `invert` and `0deg` for `hue-rotate` function?

DarknessOpen, 2022-02-28:
@jow- It's true, it disgusts me how you look at those graphs now.

You can't let them look white, if someone is using the beautiful black theme.

[image]