openwrt/luci #5692: DDNS - IPv6 is currently not (fully) supported by this system

zcy85611, 2022-02-21:
After I compiled the latest openwrt trunk and upgraded the kernel to 5.10, DDNS keeps telling me "IPv6 is currently not (fully) supported by this system, Please follow the instructions on OpenWrt's homepage to enable IPv6 support or update your system to the latest OpenWrt Release"

But I actually got a ipv6 address and has passed the ipv6 test on www.test-ipv6.com, on kernel 5.4, I did it the same way and DDNS works fine

could anyone happen to know why is it happening and help on how to solve it? thank you much!

joshenders, 2022-02-27:
Experiencing this as well on `OpenWrt SNAPSHOT, r18974-8144f9c665` which is kernel 5.10

joshenders, 2022-02-27:
Decided to look a little closer at this, it appears IPv6 support is detected in `luci/applications/luci-app-ddns/root/usr/libexec/rpcd/luci.ddns` on [line 274](https://github.com/openwrt/luci/blob/b5b66e994dff1c73a826a5dfe54bed413b9e4913/applications/luci-app-ddns/root/usr/libexec/rpcd/luci.ddns#L274) by checking for the presence of both `/proc/net/ipv6_route` and `/usr/sbin/ip6tables`.

This is later used in [luci/applications/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js](https://github.com/openwrt/luci/blob/cd2b9b40070cf55523f0f2fd6490c70952a3f754/applications/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js#L305-L312) which is what is displaying the error message.

So the next obvious thing is to look for the presence of these two files... and low and behold, the issue appears to be caused by `ip6tables` not being installed.

```
root@openwrt:~# ls -l /proc/net/ipv6_route
-r--r--r--    1 root     root             0 Feb 27 02:24 /proc/net/ipv6_route
root@openwrt:~# ls -l /usr/sbin/ip6tables
ls: /usr/sbin/ip6tables: No such file or directory
```

You can make this error message disappear by installing `ip6tables`:
```
opkg update
opkg install ip6tables
```

but I'm not sure if this is the best way to fix the underlying issue here. e.g whether this is the best way to detect for IPv6 support in the ddns package or if a recent change (some `fw3` prep?) has deprecated ip6tables usage or if `ip6tables` is not included in the snapshot for some other specific reason. Just not familiar enough with the project for qualified insight here.

Maybe somebody else can chime in?

commit 575fb0c8 (2022-03-10): luci-app-ddns: assume IPv6 support if nftables is present

Fixes: #5692
closed 2022-03-10 by jow-