openwrt/luci #5693: luci-theme-bootstrap: Buttons at Status/Overview/Associated Stations hard to use

wdpi, 2022-02-21:
## Steps to reproduce:

1. go to: Network → Wireless → Edit (one of your Wireless Networks) → Interface Configuration → MAC Filter
2. set "Allow all except listed"
3. add a MAC (probably not necessary for reproducing) in MAC-List
4. Save
5. Save & Apply

## Actual behavior:

go to: Status/Overview/Associated Stations

The Buttons on the right side of every associated station now are not just Buttons labelled "Disconnect" but a combination of Button and Combo box roughly divided into two halves by a vertical line.
The left part contains the text "Disc" while the right one is empty except a Combo box arrow close to the vertical line.

The Combo box contains the items "Disconnect" and "Add to Blacklist".

Opening the Combo box is quite hard since just a relatively small area around the arrow triggers that behaviour - clicking at the most parts of the button just disconnects immediately instead of opening the Combo box.

## Expected behavior:

A Button/Combo box layout like e.g. the "Save & Apply" Button would be clear.

## Additional Information:
luci-theme-bootstrap git-22.047.35373-cc582eb

jow-, 2022-02-21:
Please attach a screenshot

wdpi, 2022-02-21:
Done.
I did't know it was possible and so easy.

jow-, 2022-02-21:
Which version of OpenWrt is that? Did you upgrade individual LuCI components?

jow-, 2022-02-21:
Was able to reproduce it. Fix pushed.

closed 2022-02-21 by jow-

wdpi, 2022-02-21:
I have 
openwrt-21.02.1-ath79-generic-tplink_archer-c6-v2-squashfs-factory.bin 
installed.

Powered by LuCI openwrt-21.02 branch (git-21.295.67054-13df80d) / 
OpenWrt 21.02.1 r16325-88151b8303

The problem existed in this version already, so I updated to 
luci-theme-bootstrap git-22.047.35373-cc582eb - which did not solve the 
problem.

Nothing else.

commit edbde448 (2022-02-21): luci-theme-bootstrap: fix display glitch with combo buttons in row actions

Fixes: #5693
commit 2dba71e4 (2022-02-21): luci-theme-bootstrap: fix display glitch with combo buttons in row actions

Fixes: #5693

(cherry picked from commit edbde448b8fae99a5646951aef5f3717dc283c87)