openwrt/luci #5709: luci-app-attendedsysupgrade: broken

BabochkaAce, 2022-03-05:
this commit #5704 broke the luci-app-attendysupgrade package

[image]

JulioMC81, 2022-03-05:
Hi! Same problem here, snapshot r19043:
[image]

Thanks! :)

castillofrancodamian, 2022-03-09:
[image]

aparcar, 2022-03-09:
Fixed via https://github.com/openwrt/luci/commit/65266c490a40e589bde23acb1bcac401813c43ce

Sorry for the delay.

closed 2022-03-09 by aparcar

JulioMC81, 2022-03-09:
> Fixed via [65266c4](https://github.com/openwrt/luci/commit/65266c490a40e589bde23acb1bcac401813c43ce)
> 
> Sorry for the delay.

Thanks @aparcar!!!