openwrt/luci #5713: luci-app-nft-qos: rate.htm can not find anything

openips, 2022-03-09:
luci-app-nft-qos/luasrc/view/nft-qos/rate.htm

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='master'
DISTRIB_REVISION='OpenWrt SNAPSHOT r19075-ed364cd4b0 / LuCI Master git-22.058.70382-d29400e'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
kernel verion=5.10.1031
nft-qos = 1.0.6-4
luci-app-nft-qos = git-22.026.38400-f30b673

with the master git version SNAPSHOT r19075-ed364cd4b0 / LuCI Master git-22.058.70382-d29400e
visite http://192.168.1.1/cgi-bin/luci/admin/status/realtime/rate cannot see any host
check code and fix bug please
thks a lot 

[image]

A-Pallotta, 2022-06-08:
The issue seems to be caused by the fact that what is retrieved from `nft list chain ip nft-qos-monitor [download|upload]` does not have any information by default:

```
table inet nft-qos-monitor {
        chain upload {
                type filter hook postrouting priority filter; policy accept;
                ip daddr 192.168.1.249 counter packets 0 bytes 0
        }

        chain download {
                type filter hook prerouting priority filter; policy accept;
                ip saddr 192.168.1.249 counter packets 0 bytes 0
        }
}
```

The controller then does not generate the JSON array because the if statement fails:

```
if i and p and b then <--- This fails
    -- handle expression
        rv[#rv + 1] = {
	    rule = {
		    family = "inet",
		    table = "nft-qos-monitor",
		    chain = n,
		    handle = 0,
		    expr = {
		       { match = { right = i } },
		       { counter = { packets = p, bytes = b } }
		   }
	  }
     }
end
```

However, manually adding a new rule (for debugging: `nft add rule inet nft-qos-monitor [download|upload] ip daddr $IPADDR counter`)  seems to be working just fine.

[image]

ZBXallen, 2022-06-17:
I can find my host on the 'nft-qos-monitor';
table inet nft-qos-monitor {
        chain upload {
                type filter hook postrouting priority filter; policy accept;
                ip saddr 192.168.1.201 counter packets 0 bytes 0
                ip saddr 192.168.1.220 counter packets 13 bytes 1917
                ip saddr 192.168.1.214 counter packets 187 bytes 27010
        }

        chain download {
                type filter hook prerouting priority filter; policy accept;
                ip daddr 192.168.1.201 counter packets 3 bytes 408
                ip daddr 192.168.1.220 counter packets 13 bytes 1378
                ip daddr 192.168.1.214 counter packets 179 bytes 91632
        }
}
but the page doesn`t show anything 
thanks
[image]

A-Pallotta, 2022-06-23:
Does the dev console show any error? Maybe what is being sent to the frontend is not mapped correctly. If it does not show any error, try logging the response from lua

ManuLinares, 2022-08-12:
> Does the dev console show any error? Maybe what is being sent to the frontend is not mapped correctly. If it does not show any error, try logging the response from lua

the json generation is working, as json data appears in http://192.168.1.1/cgi-bin/luci/admin/status/realtime/rate_status

This is the console.log of http://192.168.1.1/cgi-bin/luci/admin/status/realtime/rate

```
Direct use XHR() is deprecated, please use L.Request instead luci.js:222:9
Uncaught TypeError: L.ui is undefined
    cbi_update_table http://192.168.1.1/luci-static/resources/cbi.js?v=git-22.175.85350-d4c4afe:133
    <anonymous> http://192.168.1.1/luci-static/resources/cbi.js?v=git-22.175.85350-d4c4afe:141
    EventListener.handleEvent* http://192.168.1.1/luci-static/resources/cbi.js?v=git-22.175.85350-d4c4afe:139
cbi.js:133:49
    cbi_update_table http://192.168.1.1/luci-static/resources/cbi.js?v=git-22.175.85350-d4c4afe:133
    forEach self-hosted:164
    <anonymous> http://192.168.1.1/luci-static/resources/cbi.js?v=git-22.175.85350-d4c4afe:141
    (Async: EventListener.handleEvent)
    <anonymous> http://192.168.1.1/luci-static/resources/cbi.js?v=git-22.175.85350-d4c4afe:139

```

After that is keeps sending and receiving the json data, but nothing shows up.

relevant files
1. https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/luci.js
1. https://github.com/openwrt/luci/blob/master/applications/luci-app-nft-qos/luasrc/view/nft-qos/rate.htm

ManuLinares, 2022-08-24:
I have a workaround.
Here is the diff, to test it.

https://github.com/openwrt/luci/compare/master...ManuLinares:luci:patch-2

I'm afraid to make a PR, as I see too many dead PR's from years ago. And a simple one character text correction I've done, isn't even getting upstream.

ManuLinares, 2022-11-09:
https://github.com/openwrt/luci/compare/master...ManuLinares:luci:master#diff-c9d9a016b1d075bce503b6cd004b701fcbc7baacc5f4acc60601ee72fc404d6b

another patch to get rate.htm working...

openips, 2022-11-10:
thks a lot

piterq, 2022-12-25:
@ManuLinares I've tested your fixed file (by replacing rate.htm file) on OpenWrt 22.03-SNAPSHOT r19977 but it's not working in my setup - "Collecting data..." text is gone, but there is no data shown:
[image]

Is it still working for you?

ManuLinares, 2022-12-25:
> @ManuLinares I've tested your fixed file (by replacing rate.htm file) on OpenWrt 22.03-SNAPSHOT r19977 but it's not working in my setup - "Collecting data..." text is gone, but there is no data shown: 
> 
> Is it still working for you?

Yes it is working, You should [update your openwrt](https://openwrt.org/docs/guide-user/installation/attended.sysupgrade) and reupload the "rate.htm"

piterq, 2022-12-26:
I'm using [snapshot build (from december 9th 2022)](https://github.com/openwrt/openwrt/commits/6cd1035d8d), so it's one of the latest. What exactly version and build are you using?

ManuLinares, 2022-12-26:
> I'm using [snapshot build (from december 9th 2022)](https://github.com/openwrt/openwrt/commits/6cd1035d8d), so it's one of the latest. What exactly version and build are you using?

Firmware Version | OpenWrt SNAPSHOT r21521-7fb32285d5 / LuCI Master git-22.325.41248-fa17c15
Kernel Version | 5.15.83

`scp -O 'rate.htm' routerip:/usr/lib/lua/luci/view/nft-qos/rate.htm`

[rate.htm.zip](https://github.com/openwrt/luci/files/10303973/rate.htm.zip)

piterq, 2022-12-26:
This is exactly the same file and the same location as I am using. How did you get the console.log file mentioned in [this comment](https://github.com/openwrt/luci/issues/5713#issuecomment-1213072571)?

piterq, 2022-12-26:
I don't know what to say. I've replaced rate.htm file once more and this time it worked :O I'm 100% sure that I've done it correctly before. Maybe there was something in browser cache...

jow-, 2022-12-27:
> Uncaught TypeError: L.ui is undefined

Should have been fixed by https://github.com/openwrt/luci/commit/bdfdd95a273f6fc73017795805dcfe6ee1ae8a8e in master and https://github.com/openwrt/luci/commit/44ecf5f3f7dc0abee0bfae4462023ebaddcac715 in openwrt-22.03. 

Please see if the issue still occurs after the fixes above.

jow-, 2022-12-27:
The issue was something else, fixed with https://github.com/openwrt/luci/commit/deed6827b2eb0b58ee8c1d810a584a9589108eac in master and  https://github.com/openwrt/luci/commit/438c598e94cd300750eaadbb3981418368812970 in openwrt-22.03

closed 2022-12-27 by jow-

ManuLinares, 2022-12-28:
indeed, this is fixed.

I uploaded https://raw.githubusercontent.com/openwrt/luci/deed6827b2eb0b58ee8c1d810a584a9589108eac/modules/luci-base/htdocs/luci-static/resources/ui.js to **/www/luci-static/resources/ui.js** and it works

`scp -O ui.js router:/www/luci-static/resources/ui.js`

piterq, 2022-12-28:
I confirm that too. I've restored original rate.htm file and I've uploaded ui.js file - everything looks good.

commit deed6827 (2022-12-27): luci-base: ui.js: support div based table markup in UITable.update()

Constructing UITable instances from existing, div based markup is
supported but the UITable.update() implementation did not account
for that, leading to defunct data updates on tables built from div
based markup.

Fix this issue by extending UITable.update() to consider a div based
table structure as well, like we do in UITable.initFromMarkup() already.

Fixes: #5713
commit 438c598e (2022-12-27): luci-base: ui.js: support div based table markup in UITable.update()

Constructing UITable instances from existing, div based markup is
supported but the UITable.update() implementation did not account
for that, leading to defunct data updates on tables built from div
based markup.

Fix this issue by extending UITable.update() to consider a div based
table structure as well, like we do in UITable.initFromMarkup() already.

Fixes: #5713

(cherry picked from commit deed6827b2eb0b58ee8c1d810a584a9589108eac)