openwrt/luci #5714: switch.js: switch page add vlan failed if 2 or more switch exist

ptpt52, 2022-03-09:
/etc/board.json
```
...
        "switch": {
                "switch0": {
                        "enable": true,
                        "reset": true,
                        "ports": [
                                {
                                        "num": 0,
                                        "role": "lan"
                                },
                                {
                                        "num": 1,
                                        "role": "lan"
                                },
                                {
                                        "num": 2,
                                        "role": "lan"
                                },
                                {
                                        "num": 3,
                                        "role": "lan"
                                },
                                {
                                        "num": 4,
                                        "role": "wan"
                                },
                                {
                                        "num": 6,
                                        "device": "eth0",
                                        "need_tag": false,
                                        "want_untag": false
                                }
                        ],
                        "roles": [
                                {
                                        "role": "lan",
                                        "ports": "0 1 2 3 6t",
                                        "device": "eth0.1"
                                },
                                {
                                        "role": "wan",
                                        "ports": "4 6t",
                                        "device": "eth0.2"
                                }
                        ]
                },
                "switch1": {
                        "enable": true,
                        "reset": true,
                        "ports": [
                                {
                                        "num": 0,
                                        "role": "lan"
                                },
                                {
                                        "num": 1,
                                        "role": "lan"
                                },
                                {
                                        "num": 2,
                                        "role": "lan"
                                },
                                {
                                        "num": 3,
                                        "role": "lan"
                                },
                                {
                                        "num": 5,
                                        "device": "eth1",
                                        "need_tag": false,
                                        "want_untag": false
                                }
                        ],
                        "roles": [
                                {
                                        "role": "lan",
                                        "ports": "0 1 2 3 5t",
                                        "device": "eth1.1"
                                }
                        ]
                }
        },

...
```

switch.js not work property if 2 or more switch dev exist

ptpt52, 2022-03-09:
[image]

ptpt52, 2022-03-11:
@jow-  hi
could you help fix this issue?

ptpt52, 2022-03-11:
switch display wrong:
config/network:
```
config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '0 1 2 3 6t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '4 6t'

config switch
        option name 'switch1'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch1'
        option vlan '1'
        option ports '0 1 2 3 5t'

```

vlan 2 missing on switch0 on the page.

jow-, 2022-05-03:
Fixed by #5718

closed 2022-05-03 by jow-