openwrt/luci #5719: luci-app-statistics - Cannot set collectd IRQs to watch through Luci

VA1DER, 2022-03-11:
## Steps to reproduce:

1. opkg install luci-app-statistics collectd-mod-irq
2. Visit Statistics -> Setup -> General Plugins, enable Interrupts
3. Click "Configure" on Interrupts, delete the default set and enter in your own set to watch
4. Click "Save", then "Save and Apply"
5. Visit Statistics -> Graphs -> Interrupts

## Actual behavior:
Some or all of the interrupts selected for monitoring in step 3 will be missing from the interrupts graph

## Expected behavior:
Requested interrupts should be present

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.2'
DISTRIB_REVISION='r16495-bf0c965af0'
DISTRIB_TARGET='mediatek/mt7623'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 21.02.2 r16495-bf0c965af0'
```

## Cause:
The cause appears to be a lack of quotes around Irq values that are being written to `/etc/collectd.conf`.  For example, when I select IRQs 205 and 206, this is what the irq plugin section looks like:

```
LoadPlugin irq
<Plugin irq>
	IgnoreSelected false
	Irq 205
	Irq 206
</Plugin>
```

And that does not work.  If I manually edit to enclose the Irq #s in quotes, as follows:

```
LoadPlugin irq
<Plugin irq>
	IgnoreSelected false
	Irq "205"
	Irq "206"
</Plugin>
```
...then it works as expected.

hnyman, 2022-03-11:
Duplicate of #4910 
Fixed in master by #5277  and that should be backported to 21.02

hnyman, 2022-03-13:
Backported as 8cac3cbc

commit 8cac3cbc (2022-03-13): luci-app-statistics: irq plugin: handle numeric interrupt names

the `irq` plugin does not show the right data if the interrupt name is only numeric

(cherry picked from commit dc2570305383a1014b39b44471c0b8c491e23026)
Fixes #5719
closed 2022-03-13 by hnyman