openwrt/luci #5731: luci-app-openvpn: openvpn config may be corrupted on upload

anyuta1166, 2022-03-19:
## Steps to reproduce:
1. Go to VPN - OpenVPN
2. upload OVPN configuration file

## Actual behavior:
Configuration file may be corrupted.

## Expected behavior:
Configuration file should be not modified on upload.

## Additional Information:

In my specific case "-----END OpenVPN **Static key** V1-----" was replaced with "-----END OpenVPN **Statickey** V1-----" and OpenVPN didn't start. I obviously can't share my config file because it contains keys. But it is reproducible on any OpenWRT build since 2018-10-23 when the ovpn file upload support was added.

anyuta1166, 2022-03-19:
Should be fixed by #5732

jow-, 2022-04-27:
Fixed by #5775 5732

closed 2022-04-27 by jow-