openwrt/luci #5734: luci: setting option 'dhcpv4' 'server' on interface results in nothing happening

it-jonas, 2022-03-20:
## Steps to reproduce:

1. go to: Network → Interfaces → LAN (an interface with a static address) → DHCP server tab → General setup tab
2. tick 'ignore interface'
3. Save & Apply
4. untick 'ignore interface'
5. Save & Apply
6. read /etc/config/network

## Actual behavior:

1. option dhcpv4 'server' in /etc/config/dhcp does not show up.
2. In LuCI 'ignore interface' is still unticked

## Expected behavior:

option gets set in /etc/config/dhcp

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.0-rc4'
DISTRIB_REVISION='r16256-2d5ee43dc6'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 21.02.0-rc4 r16256-2d5ee43dc6'
DISTRIB_TAINTS=''
```

jow-, 2022-03-20:
LuCI does not implement any `option dhcpv4`

closed 2022-03-20 by jow-

jelle-voet-aleacsysonline, 2022-04-01:
Ok, but there used to be such an option (next to the IPv6 Settings tab, there was an entire IPv4 Settings tab).

The option is still supported in the config file. Any idea why it was removed from Luci ?