openwrt/luci #5758: <LUCI-APP-FRPC>FRPC can't save new settings

Beaverfffan, 2022-04-04:
when i save new settings at frpc，message following will show out。

1. go to: Service → Interfaces → frp settings
2. set new frpc link
3. Save & Apply

4.Save error
An error occurred while saving the form:

RPC call to uci/delete failed with ubus code 4: Resource not found
at ClassConstructor.handleCallReply (http://192.168.0.1/luci-static/resources/rpc.js?v=git-22.090.69956-24ed48d:15:3
## Additional Information:
OpenWrt version information from system

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='master'
DISTRIB_REVISION='all'
DISTRIB_TARGET='all'
DISTRIB_ARCH='all'
```

Beaverfffan, 2022-04-15:
maybe  it was caused by null?it seems like some luci settings with value “null ” can't set properly.

jow-, 2022-07-23:
I believe this issue has been fixed by 2771360108cb3e2011c7c9cb4921a9047ea70910

closed 2022-07-23 by jow-