openwrt/luci #5760: luci-app-mwan3: can't add interfaces

paladin4fan, 2022-04-06:
## Steps to reproduce:

1. go to: Network → Multiwan→interfaces tab
2. delete all default interfaces here, then create a new interface, setting the tracking IP and enable it
3. Save & Apply
4. go to: interfaces setting tab again

## Actual behavior:

1. nothing shows here, the interfaces keeps empty
2. no interface information in /etc/config/mwan3

## Expected behavior:

```
Correctly show the interface message in both luci and config file
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r19364-6af46796fa'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r19364-6af46796fa'
DISTRIB_TAINTS=''
```

feckert, 2022-06-21:
I can reproduce the error.

However, if I use `save&apply` from the uci change overview then it works.

[image]

Strange?

openwrtdiy, 2022-06-23:
Also found a new issue: When setting the assigned policy: wan_wanb, the network failover did not take effect. After uploading and installing MWAN3_2.8.16, the policy will take effect.

feckert, 2022-07-12:
Only the openwrt-21.02 branch works as expected because this is still using the lua facility.

The master and openwrt-22.03 branch does not work as expected!

I have changed the luci-app-wan3 so that it now uses javascript in this branches.
But I am sure that everything already worked after javascript migration.
Somewhere a regression slipt in.

Interesting is, that I can't edit the section after creating it. I am getting a empty modal.
[image]
But if I go into another luci-app-* and back again, then I can edit it.

Very funny behaviour.  @jow- can you help us there?
I have already looked at the history but have not seen any suspicious commit.
I have to mention that when I go to the modal **Unsaved changes** and click on `Save & Apply`, the configuration gets applied and will be saved!

[image]

paladin4fan, 2022-07-17:
> That muss be a generic problem. Could you verify if this is als

I've compiled from 22.03.0-RC5 and tried submit through uci change overview
And it works.

paladin4fan, 2022-07-17:
> I can reproduce the error.
> 
> However, if I use `save&apply` from the uci change overview then it works.
> 
> [image]
> 
> Strange?
> 
> @paladin4fan Could you verify if this is the same on your system?
> 
> That muss be a generic problem. Could you verify if this is also not working for application that uses the same Luci template [gridsection](https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/interface.js#L26)

BTW in menber tabs it got a new issue
I can't add member continuously. If I add a new member follow the former member without submit from uci changes overview, the hinder member will cover the former one.

feckert, 2022-07-21:
I use the Luci pattern here. So nothing special. This should also be the case with other apps. Can you confirm this?

jow-, 2022-07-22:
It is indeed a generic LuCI issue. Just tracked down the problem but still thinking about a proper solution.

closed 2022-07-22 by jow-

1715173329, 2022-09-07:
wonder if 21.02 branch also needs this fix... form.js looks pretty old there

commit 27713601 (2022-07-22): luci-base: form.js: fix removing just created named GridSection

Since the `handleModalSave()` handler of the GridSection class invokes
`handleModalCancel()` after saving the data but before removing the
`addedSection` property, the `handleModalCancel` handler incorrectly
removed the uci section that has just been created.

This bug didn't affect anonymous GridSections because after saving the
id of the created section changes, causing the remove command to fail,
but for named ones with stable section IDs, the bug manifested.

Solve the issue by passing a flag to `handleModalCancel()` indicati [... cut]
commit 1e7d9e44 (2022-07-22): luci-base: form.js: fix removing just created named GridSection

Since the `handleModalSave()` handler of the GridSection class invokes
`handleModalCancel()` after saving the data but before removing the
`addedSection` property, the `handleModalCancel` handler incorrectly
removed the uci section that has just been created.

This bug didn't affect anonymous GridSections because after saving the
id of the created section changes, causing the remove command to fail,
but for named ones with stable section IDs, the bug manifested.

Solve the issue by passing a flag to `handleModalCancel()` indicati [... cut]