openwrt/luci #5768: X5000R 5GHz radio showing incorrect Wifi MAC address

liukuohao, 2022-04-17:
I bought 2 X5000R routers, and all units are showing the same MAC address in the 5GHz radio in Luci Wireless Overview.

[image]

[image]

liukuohao, 2022-04-23:
Basically ALL the versions of 21.02.X & also the latest branch 22.03.0

Another issue is 21.02.3 firmware still does not work well with 5GHz radio. 
It will cause to have TypeError red message pop out when you are saving
settings. In order to solve this entirely, only the new branch 22.03.0 works

jow-, 2022-04-27:
This is caused by a too old libiwinfo in OpenWrt base which does not understand the new `option path ...+1` syntax used to identify the 5GHz radio phy. This has all sorts of weird side effects, like not being to identify the available operation modes, frequencies and so on.

I backported libiwinfo updates into the openwrt-21.02 branch now, the autobuilders will pick those up eventually and push new package versions to the binary repos. To fix the issue, you can then upgrade `libiwinfo20210430` via opkg.

OpenWrt 21.02.4 will contain the updated package directly then.

closed 2022-04-27 by jow-

liukuohao, 2022-04-27:
**THANK YOU** for fixing the problem.