openwrt/luci #5771: [openwrt/luci] Changing disassoc_low_ack wireless settings breaks 5g wireless [X5000R]

freaksdotcom, 2022-04-20:
# Issue template:
1. Please make sure that the issue subject starts with `<package-name>:`
2. Issues related to end of life (EOL) releases are not supported or maintained and will be closed
3. Remove lines from top till here and fill in the following questions
***

## Steps to reproduce:
1. Network -> Wireless -> Edit a 5g SSID
2. Interface Configuration -> Advanced
3. Uncheck 'Disassociate On Low Acknowledgement'
4. Save & apply

## Actual behavior:
Changes show the following:
# /etc/config/wireless
uci del wireless.radio1.band
uci del wireless.radio1.channel
uci add wireless.wifinet4.disassoc_low_ack='0'

Applying the changes disables wireless.

## Expected behavior:
**Example: Replace the following lines and remove this**
`wireless.radio1.band` and `wireless.radio1.channel` should not be deleted.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
root@Tvroom:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.2'
DISTRIB_REVISION='r16495-bf0c965af0'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.2 r16495-bf0c965af0'
DISTRIB_TAINTS=''
```

```
root@Tvroom:~# opkg list-installed | grep luci
liblucihttp-lua - 2021-06-11-3dc89af4-1
liblucihttp0 - 2021-06-11-3dc89af4-1
luci - git-20.074.84698-ead5e81
luci-app-acl - git-21.194.67638-1d6053e
luci-app-firewall - git-22.089.67741-3856d50
luci-app-opkg - git-21.312.69848-4745991
luci-base - git-22.083.69138-0a0ce2a
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-22.097.61937-bc85ba5
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.046.85061-dd54dce
luci-mod-status - git-22.046.85784-0ac2542
luci-mod-system - git-22.019.40321-7a37d02
luci-proto-ipv6 - git-21.148.49484-14511e5
luci-proto-ppp - git-21.163.64918-6c6559a
luci-ssl - git-20.244.36115-e10f954
luci-theme-bootstrap - git-22.084.39047-f1d687e
rpcd-mod-luci - 20210614
```

jow-, 2022-04-21:
Same issue as #5468

freaksdotcom, 2022-04-21:
It's not the same. #5468 is fixed in the version of Luci I'm running. 

It may be the same root cause, but it is not fixed.

jow-, 2022-04-21:
LuCI in 21.02 does not properly support this model, not all required changes have been backported.

freaksdotcom, 2022-04-21:
This is on nightlies.

jow-, 2022-04-21:
> DISTRIB_RELEASE='21.02.2'

freaksdotcom, 2022-04-26:
This is still happening on 21.02.03.

```
root@den:/etc/config# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
root@den:/etc/config# opkg list-installed | grep luci
liblucihttp-lua - 2021-06-11-3dc89af4-1
liblucihttp0 - 2021-06-11-3dc89af4-1
luci - git-20.074.84698-ead5e81
luci-app-firewall - git-22.089.67741-3856d50
luci-app-opkg - git-21.312.69848-4745991
luci-base - git-22.083.69138-0a0ce2a
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-22.097.61937-bc85ba5
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.046.85061-dd54dce
luci-mod-status - git-22.046.85784-0ac2542
luci-mod-system - git-22.019.40321-7a37d02
luci-proto-ipv6 - git-21.148.49484-14511e5
luci-proto-ppp - git-21.163.64918-6c6559a
luci-ssl - git-20.244.36115-e10f954
luci-theme-bootstrap - git-22.084.39047-f1d687e
rpcd-mod-luci - 20210614
```

jow-, 2022-04-26:
> This is still happening on 21.02.03.

The 21.02.3 branch still lacks crucial backports

freaksdotcom, 2022-04-26:
The bug isn't fixed in the current release. It is still a problem.

Upgrading to a nightly is not possible due to the missing backports. This leaves users of the "production" release unable to use the software.

jow-, 2022-04-26:
Please test if a current snapshot version of OpenWrt works. Also provide the complete output of the following commands:

 - `uci show wireless`
 - `ubus call iwinfo info '{ "device": "radio1" }'`

Additionally, provide a screenshot of the wireless configuration screen before the change and after saving the change.

jow-, 2022-04-27:
This is caused by a too old libiwinfo in OpenWrt base which does not understand the new `option path ...+1` syntax used to identify the 5GHz radio phy. This has all sorts of weird side effects, like not being to identify the available operation modes, frequencies and so on.

I backported libiwinfo updates into the openwrt-21.02 branch now, the autobuilders will pick those up eventually and push new package versions to the binary repos. To fix the issue, you can then upgrade `libiwinfo20210430` via opkg.

OpenWrt 21.02.4 will contain the updated package directly then.

closed 2022-04-27 by jow-

freaksdotcom, 2022-04-28:
> Please test if a current snapshot version of OpenWrt works. Also provide the complete output of the following commands:
```
# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r19537-c283defa88'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r19537-c283defa88'
DISTRIB_TAINTS=''
```

```
# opkg list-installed|grep luci
liblucihttp-lua - 2022-02-13-cc851838-1
liblucihttp0 - 2022-02-13-cc851838-1
luci - git-20.074.84698-ead5e81
luci-app-firewall - git-22.089.67453-0eb3aeb
luci-app-opkg - git-22.043.53495-1e6f630
luci-base - git-22.117.40763-2e4b5fb
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-22.079.54693-45f411b
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.117.67385-1ae142d
luci-mod-status - git-22.117.40788-2f80fe3
luci-mod-system - git-22.117.68098-c2eb30b
luci-proto-ipv6 - git-21.148.48881-79947af
luci-proto-ppp - git-21.158.38888-88b9d84
luci-theme-bootstrap - git-22.084.38997-8e09f18
rpcd-mod-luci - 20210614
```

```
# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
wireless.radio0.band='2g'
wireless.radio0.cell_density='0'
wireless.radio0.country='US'
wireless.radio0.htmode='HE40'
wireless.radio0.channel='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.path='1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
wireless.radio1.band='5g'
wireless.radio1.cell_density='0'
wireless.radio1.htmode='HE80'
wireless.radio1.country='US'
wireless.radio1.distance='5'
wireless.radio1.channel='auto'
wireless.wifinet1=wifi-iface
wireless.wifinet1.ssid='XXXXXX'
wireless.wifinet1.encryption='psk2'
wireless.wifinet1.device='radio0'
wireless.wifinet1.mode='ap'
wireless.wifinet1.network='lan'
wireless.wifinet1.key='YYYYYYYYYYYY'
wireless.wifinet1.disabled='1'
wireless.wifinet1.disassoc_low_ack='0'
wireless.wifinet1.ieee80211r='1'
wireless.wifinet1.ft_over_ds='1'
wireless.wifinet1.ft_psk_generate_local='1'
wireless.wifinet2=wifi-iface
wireless.wifinet2.ssid='XXXXXX-guest'
wireless.wifinet2.encryption='psk2'
wireless.wifinet2.device='radio0'
wireless.wifinet2.mode='ap'
[... 255 lines cut ...]
			"mhz": 5805,
			"restricted": false,
			"active": false
		},
		{
			"channel": 165,
			"mhz": 5825,
			"restricted": false,
			"active": false
		},
		{
			"channel": 169,
			"mhz": 5845,
			"restricted": true,
			"active": false
		},
		{
			"channel": 173,
			"mhz": 5865,
			"restricted": true,
			"active": false
		}
	]
}
```

freaksdotcom, 2022-05-02:
21.02.3 with upgraded packages is working.

```
# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
```

```
# opkg list-installed|grep luci
liblucihttp-lua - 2021-06-11-3dc89af4-1
liblucihttp0 - 2021-06-11-3dc89af4-1
luci - git-20.074.84698-ead5e81
luci-app-firewall - git-22.089.67741-3856d50
luci-app-opkg - git-21.312.69848-4745991
luci-base - git-22.119.37126-a993714
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-22.097.61937-bc85ba5
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.046.85061-dd54dce
luci-mod-status - git-22.046.85784-0ac2542
luci-mod-system - git-22.019.40321-7a37d02
luci-proto-ipv6 - git-21.148.49484-14511e5
luci-proto-ppp - git-21.163.64918-6c6559a
luci-ssl - git-20.244.36115-e10f954
luci-theme-bootstrap - git-22.084.39047-f1d687e
rpcd-mod-luci - 20210614
```

```
# ubus call iwinfo info '{ "device": "radio1" }'

{
	"phy": "phy1",
	"bssid": "00:0C:43:4A:21:50",
	"country": "US",
	"mode": "Master",
	"channel": 52,
	"center_chan1": 58,
	"frequency": 5260,
	"frequency_offset": 0,
	"txpower": 23,
	"txpower_offset": 0,
	"quality_max": 70,
	"noise": 0,
	"htmodes": [
		"HT20",
		"HT40",
		"VHT20",
		"VHT40",
		"VHT80",
		"HE20",
		"HE40",
		"HE80",
		"HE80+80",
		"HE160"
	],
	"hwmodes": [
		"ac",
		"ax",
		"n"
	],
	"hwmode": "ac",
	"htmode": "VHT80",
	"hardware": {
		"id": [
			5315,
			30997,
			5315,
			30997
		],
		"name": "MediaTek MT7915E"
	}
}
```

```
# ubus call iwinfo freqlist '{ "device": "radio1" }'
{
	"results": [
		{
			"channel": 36,
			"mhz": 5180,
			"restricted": false,
			"active": false
		},
		{
			"channel": 40,
			"mhz": 5200,
			"restricted": false,
			"active": false
		},
		{
			"channel": 44,
			"mhz": 5220,
			"restricted": false,
			"active": false
		},
		{
			"channel": 48,
			"mhz": 5240,
			"restricted": false,
			"active": false
		},
		{
			"channel": 52,
			"mhz": 5260,
			"restricted": false,
			"active": true
		},
		{
			"channel": 56,
			"mhz": 5280,
			"restricted": false,
			"active": false
		},
		{
			"channel": 60,
			"mhz": 5300,
			"restricted": false,
			"active": false
		},
		{
			"channel": 64,
			"mhz": 5320,
			"restricted": false,
			"active": false
		},
		{
			"channel": 100,
			"mhz": 5500,
			"restricted": false,
			"active": false
		},
		{
			"channel": 104,
			"mhz": 5520,
			"restricted": false,
			"active": false
		},
		{
			"channel": 108,
			"mhz": 5540,
			"restricted": false,
			"active": false
		},
		{
			"channel": 112,
			"mhz": 5560,
			"restricted": false,
			"active": false
		},
		{
			"channel": 116,
			"mhz": 5580,
			"restricted": false,
			"active": false
		},
		{
			"channel": 120,
			"mhz": 5600,
			"restricted": false,
			"active": false
		},
		{
			"channel": 124,
			"mhz": 5620,
			"restricted": false,
			"active": false
		},
		{
			"channel": 128,
			"mhz": 5640,
			"restricted": false,
			"active": false
		},
		{
			"channel": 132,
			"mhz": 5660,
			"restricted": false,
			"active": false
		},
		{
			"channel": 136,
			"mhz": 5680,
			"restricted": false,
			"active": false
		},
		{
			"channel": 140,
			"mhz": 5700,
			"restricted": false,
			"active": false
		},
		{
			"channel": 144,
			"mhz": 5720,
			"restricted": false,
			"active": false
		},
		{
			"channel": 149,
			"mhz": 5745,
			"restricted": false,
			"active": false
		},
		{
			"channel": 153,
			"mhz": 5765,
			"restricted": false,
			"active": false
		},
		{
			"channel": 157,
			"mhz": 5785,
			"restricted": false,
			"active": false
		},
		{
			"channel": 161,
			"mhz": 5805,
			"restricted": false,
			"active": false
		},
		{
			"channel": 165,
			"mhz": 5825,
			"restricted": false,
			"active": false
		},
		{
			"channel": 169,
			"mhz": 5845,
			"restricted": true,
			"active": false
		},
		{
			"channel": 173,
			"mhz": 5865,
			"restricted": true,
			"active": false
		}
	]
}
```