openwrt/luci #5777: luci-app-statistics: collectd-mod-memory preferences not saved correctly

kirenida, 2022-04-24:
## Steps to reproduce:

1. install luci-app-statistics
2. go to **Statistics** -> **Setup** -> **General Plugins**, click on **Configure...** for the **Memory** plugin
3. enable plugin and select **Absolute values**
4. **Save and Apply**
5. check saved settings in /etc/config/luci_statistics, see that there is no **_ValuesAbsolute_** set under **_config statistics 'collectd_memory'_**
6. go back to **Memory** plugin configuration, enable **Percent values**, **Save and Apply**
7. see that now we have both _**option ValuesAbsolute '0' option ValuesPercentage '1'**_ in _/etc/conf/luci_statistics_
8. go back to **Memory** plugin configuration, disable **Percent values**, enable **Absolute values**, save and apply
9. see that now both settings are missing from _/etc/conf/luci_statistics_

## Actual behavior:

Can't set Memory plugin to use Absolute values through LuCi, only Percentage values. 

## Expected behavior:

Correct configuration saved to /etc/conf/luci_statistics

## Additional Information:

root@OpenWrt:/etc# cat openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r19511-c59704334c'
DISTRIB_TARGET='ipq40xx/mikrotik'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r19511-c59704334c'
DISTRIB_TAINTS=''

jow-, 2022-04-24:
Options are usually removed from the config if they're matching their respective defaults. The `ValuesAbsolute` option [defaults to true](https://github.com/collectd/collectd/blob/main/src/memory.c#L109) - that's why it is removed when enabled.

kirenida, 2022-04-24:
> Options are usually removed from the config if they're matching their respective defaults. The `ValuesAbsolute` option [defaults to true](https://github.com/collectd/collectd/blob/main/src/memory.c#L109) - that's why it is removed when enabled.

Ok, but then why isn't ValuesAbsolute set to true in /etc/collectd.conf with the default settings?

After disadbling and then enabling the memory plugin using LuCi, setting Absolute values there and restarting collectd, I have:

/etc/config/luci_statistics:
`config statistics 'collectd_memory'
        option enable '1'
`
and

/etc/collectd.conf
`LoadPlugin memory
<Plugin memory>
        ValuesPercentage false
        ValuesAbsolute false
</Plugin>
`

Am I missing some setting somewhere? Something else I could check?

Is this a bug with collectd then?

jow-, 2022-04-24:
Yeah right, seems the config generator is wrong

kofec, 2022-04-24:
@kirenida thanks for report it. I also observe this issue. I always use workaround to set it manually. This same issue is for CPU if you want to have both absolute and percentage

jow-, 2022-04-24:
Problem is that the forms declare (flag) defaults that match the underlying collectd implementation defaults but the uci-to-collectd.conf translator (stat-genconfig) is not aware of these defaults, it'll always generate `false` options instead of omitting them if the corresponding uci option is unset.

There's various ways to solve this, with increasing invasiveness:
1) add a `o.forcewrite = true` to any `Flag` option declaring a non-standard default (`o.default = o.enabled`)
2) modify `stat-genconfig` to not emit `false` bool options for absent uci options
3) rework the plugin JSON option declarations and `stat-genconfig` to be able to encode default values

A potential solution for 2) would be the following:
```diff
diff --git a/applications/luci-app-statistics/root/usr/bin/stat-genconfig b/applications/luci-app-statistics/root/usr/bin/stat-genconfig
index fa59edc3a9..15e11e193e 100755
--- a/applications/luci-app-statistics/root/usr/bin/stat-genconfig
+++ b/applications/luci-app-statistics/root/usr/bin/stat-genconfig
@@ -235,13 +235,13 @@ function _bool( s, n, nopad )
        local pad = ""
        if not nopad then pad = "\t" end
 
-       if s and s == "1" then
-               str = pad .. n .. " true"
-       else
-               str = pad .. n .. " false"
+       if s == "1" then
+               str = pad .. n .. " true\n"
+       elseif s == "0" then
+               str = pad .. n .. " false\n"
        end
 
-       return str .. "\n"
+       return str
 end
 
 function _string( s, n, nopad )
```

Maybe you can give it a spin and see if it solves your issues.

kirenida, 2022-04-25:
> modify `stat-genconfig` to not emit `false` bool options for absent uci options

I modified `stat-genconfig` with your changes, but I still get the same result - no ValuesAbsolute in its output.

jow-, 2022-04-25:
> no ValuesAbsolute in its output.

Which was the goal. When no `option ValuesAbsolute` is in uci or when `option ValuesAbsolute 1` is set, then no `ValuesAbsolute` should be emitted (the memory plugin of collectd internally defaults to true if the option is absent). Only if `option ValuesAbsolute 0` is set in uci, the config generator shall emit `ValuesAbsolute false`.

kirenida, 2022-04-25:
> Which was the goal. When no `option ValuesAbsolute` is in uci or when `option ValuesAbsolute 1` is set, then no `ValuesAbsolute` should be emitted (the memory plugin of collectd internally defaults [...]

My bad, I'm a little out of my depth here, I was expecting the option to be present in the conf file.

Yes, I can confirm that the memory plugin is logging with the default settings (Enable / Absolute values), and with no ValuesAbsolute option in the .conf file.

jow-, 2022-04-25:
Thanks for confirming! I pushed the fix to all relevant branches now.

commit f16037ce (2022-04-25): luci-app-statistics: stat-genconfig: don't emit bools for absent uci options

Do not emit native collectd.conf boolean false options if the corresponding
uci option is unset in order to honour collectd's implicit defaults.

This fixes certain options, such as the memory plugins `ValuesAbsolute`,
being without effect if declared as default-enabled in the related plugin
uci forms.

Fixes: #5777
commit 46794ea2 (2022-04-25): luci-app-statistics: stat-genconfig: don't emit bools for absent uci options

Do not emit native collectd.conf boolean false options if the corresponding
uci option is unset in order to honour collectd's implicit defaults.

This fixes certain options, such as the memory plugins `ValuesAbsolute`,
being without effect if declared as default-enabled in the related plugin
uci forms.

Fixes: #5777

(cherry picked from commit f16037cec5e3f65cffe2f5788c264f1912435b0e)
commit 0473e99f (2022-04-25): luci-app-statistics: stat-genconfig: don't emit bools for absent uci options

Do not emit native collectd.conf boolean false options if the corresponding
uci option is unset in order to honour collectd's implicit defaults.

This fixes certain options, such as the memory plugins `ValuesAbsolute`,
being without effect if declared as default-enabled in the related plugin
uci forms.

Fixes: #5777

(cherry picked from commit f16037cec5e3f65cffe2f5788c264f1912435b0e)
closed 2022-04-25 by jow-