openwrt/luci #5780: luci-app-firewall: two entries in 22.03.0-rc1 menu

jtkohl, 2022-04-26:
## Steps to reproduce:

1. Install 22.03.0-rc1
2. Go to the Status menu
3. Notice two entries: Firewall (iptables) and Firewall (nftables)

## Actual behavior:

Two entries in the menu

## Expected behavior:

Should be just one entry, for nftables, since this release has switched to nftables?

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.0-rc1'
DISTRIB_REVISION='r19302-df622768da'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt 22.03.0-rc1 r19302-df622768da'
DISTRIB_TAINTS=''
```

I have an image-builder release built with these packages for `PROFILE=netgear_r7800`
```
 bind-dig
 bind-rndc
 bind-server-filter-aaaa
 bind-server
 block-mount
 collectd-mod-ping
 collectd-mod-lua
 collectd-mod-write-graphite
 collectd-mod-rrdtool
 diffutils
 e2fsprogs
 f2fs-tools
 gdisk
 tc-full
 iperf3
 irqbalance
 kmod-fs-ext4
 kmod-fs-f2fs
 kmod-nft-nat6
 kmod-usb-storage
 kmod-usb-storage-uas
 luci
 luci-theme-openwrt-2020
 luci-theme-bootstrap
 luci-app-ddns
 luci-app-sqm
 luci-app-statistics
 luci-app-uhttpd
 luci-app-watchcat
 luci-app-wireguard
 luci-proto-ipv6
 luci-proto-wireguard
 qrencode
 6in4
 nano
 ntp-utils
 openssl-util
 shadow-groupadd
 shadow-groupmod
 shadow-useradd
 shadow-usermod
 wireguard-tools
 tcpdump
 usbutils
 wget-ssl
 avahi-utils
 avahi-daemon-service-ssh
 avahi-daemon-service-http
 socat
```

hnyman, 2022-04-26:
So far it is intentional that both menu items (iptables and nftables page) are there.

jow-, 2022-04-26:
The iptables status is shown as long as either an `iptables` or `ip6tables` executable is present on the system as they could be thoretically used to stage iptables rule in parallel to nftables ones.

jtkohl, 2022-04-26:
> The iptables status is shown as long as either an `iptables` or `ip6tables` executable is present on the system as they could be thoretically used to stage iptables rule in parallel to nftables ones.

Ah, so that explains it.  I have SQM installed, which pulls in iptables-nft.

jow-, 2022-04-27:
I'm not overly happy with the way it is right now either. Maybe we should only display the nftables status page in the menu if `nft` is present, then add a warning banner at the top stating that we found legacy iptables rules in case there are indeed iptables rules on the system, and offer the user a link to the old iptables status page from there.

hnyman, 2022-04-27:
A more complete move toward nft-only would be better. So hiding the iptables page from menu sounds goog.

(We are running into into similar compromise on various fronts, e.g. also regarding package compatibility with nft,  iptables-nft and iptables-legacy  and the dependency matrix cause by those.)

jow-, 2022-04-27:
Fixed with 2e4b5fb8ff, 2f80fe3767

[image]
[image]

closed 2022-04-27 by jow-