openwrt/luci #5781: [openwrt/luci] Changing disassoc_low_ack wireless settings breaks 5g wireless [X5000R]

freaksdotcom, 2022-04-26:
## Steps to reproduce:
1. Network -> Wireless -> Edit a 5g SSID
2. Interface Configuration -> Advanced
3. Uncheck 'Disassociate On Low Acknowledgement'
4. Save & apply

## Actual behavior:
Changes show the following:
```
# /etc/config/wireless
uci del wireless.radio1.band
uci del wireless.radio1.channel
uci add wireless.wifinet4.disassoc_low_ack='0'
```

Applying the changes disables wireless. Trying to edit the network again gives an error 
```
TypeError
Cannot read properties of undefined (reading 'value')
```

This may be related to #5468. This is the same as #5771, which was closed without verification or a fix.

## Expected behavior:
**Example: Replace the following lines and remove this**
`wireless.radio1.band` and `wireless.radio1.channel` should not be deleted.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```

This is still happening on 21.02.03.

root@den:/etc/config# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''
```

```
root@den:/etc/config# opkg list-installed | grep luci
liblucihttp-lua - 2021-06-11-3dc89af4-1
liblucihttp0 - 2021-06-11-3dc89af4-1
luci - git-20.074.84698-ead5e81
luci-app-firewall - git-22.089.67741-3856d50
luci-app-opkg - git-21.312.69848-4745991
luci-base - git-22.083.69138-0a0ce2a
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-22.097.61937-bc85ba5
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-22.046.85061-dd54dce
luci-mod-status - git-22.046.85784-0ac2542
luci-mod-system - git-22.019.40321-7a37d02
luci-proto-ipv6 - git-21.148.49484-14511e5
luci-proto-ppp - git-21.163.64918-6c6559a
luci-ssl - git-20.244.36115-e10f954
luci-theme-bootstrap - git-22.084.39047-f1d687e
rpcd-mod-luci - 20210614
```

jow-, 2022-04-26:
Duplicate of #5771

closed 2022-04-26 by jow-

freaksdotcom, 2022-04-26:
Why did you close this and #5771 ? The problem still exists. #5771 was closed even though the problem is valid and exists in 21.02.03.