openwrt/luci #5797: luci-app-mosquitto: add anonymous and passwords file option

cconde, 2022-05-03:
This is just an improvement suggestion.

I think current web only allows for certificates. 

It would be great to have "allow_anonymous" and "password_file" global settings.

I'm not speaking of a user management in the interface, only the ability to enable and disable anonymous login and also to point to a mosquitto password file (not creating it, only using).

Thank you very much for your great work!

cconde, 2022-05-03:
I've managed to do it in LUA by adding this to the "mosquitto" section.
```
OptionalFlag(s, "allow_anonymous",  _("Allow anonymous access to this broker"), _("Allow connection to the broke bro>
o = s:option(Value, "password_file", _("Mosquitto password file"))
o.datatype = "file"
o.optional = true
```

Also I've found I need to do this change to make the persistence section work for me (running 21.02):
```
p = s:option(Flag, "persistence", _("Persistence enabled"), _("Should persistence to disk be enabled at all"))
p.rmempty = false
```

Hope this helps!

BlueWings172, 2022-05-22:
@cconde 
Thanks for posting this. I installed luci-app-mosquitto on my openwrt router and was wondering how to enable anonymous access. I am new to all this so it was very confusing and I thought I was missing something until I saw your post. I am not sure what is the purpose of the web interface if you cannot the change most important settings like anonymous access and password. There is also no documentation about the interface or it's limitations.

I have to find out how to replicate your solution as I'm not familiar with the steps that precede changing the settings you mentioned.

BlueWings172, 2022-05-22:
Solve it using steps in this page. There is a video too:

https://awanggacompnet.blogspot.com/2022/01/tutorial-install-mqqt-broker-on-openwrt.html?m=1

zissouz, 2023-12-02:
If you just want to enable anonymous access for devices this was the 'quick' method that worked for me...

Open the file ( /etc/config/mosquitto) - i used winSCP

Make the file look like this:
-------------------------------------------
```
config owrt 'owrt'
	option use_uci '1'

config mosquitto 'mosquitto'
	option log_dest 'syslog'
	option allow_anonymous '1'  

config persistence 'persistence'
	option persistence '0'

config listener
	option port '1883'

```
----------------------------------------------

It also made the option available in luci once set. If you are having problems then you can see the config your change is generating in /tmp/mosquitto.generated.conf and compare that with mosquitto documentation to give you ideas about where any errors are.