openwrt/luci #5799: Missing elements/corrupt UI in LuCi

Hew-ux, 2022-05-04:
## Issue:
After performing a sysupgrade to 21.02.3, many UI elements in LuCi have gone missing/unrendered, resulting in a buggy appearance.

The login screen:
[image]

The status screen:
[image]

During package installation, text becomes unreadable (this was the point that I understood this is definitely a bug, not a new minimalist UI direction):
[image]

## Steps to reproduce:
 1. Upgrade to 21.02.3
 2. Open LuCi using a browser such as Pale Moon.

## Actual behavior:
The UI appears broken (but still usable).

## Expected behavior:
The UI appears as intended.

## Additional Information:
Behaviour seems inconsistent between browsers, with other elements reappearing. Pale Moon in particular seems to have coped the brunt of it.

jow-, 2022-05-04:
Did you try to clear the browser cache?

jow-, 2022-05-04:
Seems to be due to the fact that PaleMoon does not support CSS `calc()` expressions in color functions such as `rgba()`, `hsl()` etc. See also https://repo.palemoon.org/MoonchildProductions/UXP/issues/1826

closed 2022-05-04 by jow-

Hew-ux, 2022-05-05:
Yes, clearing the cache did not solve the issue.

Using another theme, everything appears fine so thank you for that suggestion. In the meantime, maybe the default theme should not use features that may not be supported by all browsers yet? That's more a subjective topic though, as far as I'm concerned this will no longer be an issue once Pale Moon adds support for it.