openwrt/luci #5800: Luci cannot add new wireless network 22.03.0-rc1

Mijzelf, 2022-05-04:
On a ZyXEL P-2812HNU-F1, lantiq/xrx200, when trying to add a new wireless network in Luci in 22.03.0-rc1, it fails with error 'TypeError chan.options[0] is undefined'.

## Steps to reproduce:
In Network->Wireless remove all networks. Then try to add one.

## Actual behavior:
A red window pops up with 'TypeError chan.options[0] is undefined'

## Expected behavior:
A network added.

## Additional Information:
On this box the clean installation doesn't have a radio. A line 'echo 1 >/sys/bus/pci/rescan' in /etc/rc.local is necessary. Maybe this causes also this behavior. On 20.02 this line was also needed, and there Luci didn't have this problem.

jow-, 2022-05-04:
Please provide the output of `ubus call iwinfo info '{ "device": "radio0" }'` and `ubus call iwinfo freqlist '{ "device": "radio0" }'`

Mijzelf, 2022-05-05:
The requested output

```
root@OpenWrt:~# ubus call iwinfo info '{ "device": "radio0" }'
{
	"phy": "phy0",
	"country": "NL",
	"quality_max": 70,
	"noise": 0,
	"htmodes": [
		"HT20",
		"HT40"
	],
	"hwmodes": [
		"b",
		"g",
		"n"
	],
	"hardware": {
		"id": [
			6164,
			12386,
			6164,
			12386
		],
		"name": "Generic MAC80211"
	}
}

```
```
root@OpenWrt:~# ubus call iwinfo freqlist '{ "device": "radio0" }'
{
	"results": [
		{
			"channel": 1,
			"mhz": 2412,
			"restricted": false
		},
		{
			"channel": 2,
			"mhz": 2417,
			"restricted": false
		},
		{
			"channel": 3,
			"mhz": 2422,
			"restricted": false
		},
		{
			"channel": 4,
			"mhz": 2427,
			"restricted": false
		},
		{
			"channel": 5,
			"mhz": 2432,
			"restricted": false
		},
		{
			"channel": 6,
			"mhz": 2437,
			"restricted": false
		},
		{
			"channel": 7,
			"mhz": 2442,
			"restricted": false
		},
		{
			"channel": 8,
			"mhz": 2447,
			"restricted": false
		},
		{
			"channel": 9,
			"mhz": 2452,
			"restricted": false
		},
		{
			"channel": 10,
			"mhz": 2457,
			"restricted": false
		},
		{
			"channel": 11,
			"mhz": 2462,
			"restricted": false
		},
		{
			"channel": 12,
			"mhz": 2467,
			"restricted": false
		},
		{
			"channel": 13,
			"mhz": 2472,
			"restricted": false
		}
	]
}

```
I had to reflash my router, and now I can't reproduce the issue. But something is wrong with the wifi, 'iw wlan0 scan' doesn't give an output, nor does the scan button in Network->Wireless, nor does Status->Channel analysis. My laptop can see the AP, but can't connect. So I think something is wrong with wireless, and Luci won't have much to do with it.

closed 2022-05-10 by Mijzelf