openwrt/luci #5807: luci: no related rpc reply or getting 0 in browser on web page

WeSTManCoder, 2022-05-11:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r19620-d9a063c396'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r19620-d9a063c396'
DISTRIB_TAINTS='no-all busybox'

Sometimes I get an error no related rpc reply. I tried different browsers - the same, the connection is stable LAN. There is also a page with 0 and nothing else. How to solve this problem?

[image]

WeSTManCoder, 2022-05-11:
> Something is interfering with the connection to the router. Do you have any antivirus appliances or similar running on your local system?

DrWeb
Windows firewall is disabled

jow-, 2022-05-11:
Please ensure that it does not interfere with the connection to the router, by adblocking/ anti-phising etc. heuristics.

WeSTManCoder, 2022-05-11:
> Please ensure that it does not interfere with the connection to the router, by adblocking/ anti-phising etc. heuristics.

installed adguard, but adguard for 192.168.1.1 is disabled

Sithuk, 2022-09-27:
I have experienced this too (22.03).

Gexar, 2025-04-06:
Since it is open, I am going to chime in and say that I had a similar issue just this morning while trying to access
http://192.168.8.1/cgi-bin/luci/

the solution was to switch to
http://192.168.8.1:8080/cgi-bin/luci/

which is available as a button "Go to LuCI" via the left menu "system" -> "advanced settings" or
http://192.168.8.1/#/advanced