openwrt/luci #5809: luci-app-attendedsysupgrade: allow to install same version

CAFxX, 2022-05-11:
Note: this is an enhancement request

## Steps to reproduce:

1. Open the app and check for available updates

## Actual behavior:

Current version is not presented as an option

## Expected behavior:

Current version is presented as an option if the set of currently installed packages differs from the list of packages in the current base image.

## Additional Information:

This would allow users to rebuild the image of the current version with a different set of installed packages, something that would be especially useful if the user has a small overlay FS as it would provide a good path for users to free up space in the overlay by moving the packages stored in the overlay to the base image.

McGiverGim, 2022-05-11:
It will help too if you need to do a hard reset, to not loose the packages installed...

aparcar, 2022-06-07:
Please try https://github.com/openwrt/luci/pull/5833

CAFxX, 2022-06-24:
@aparcar sorry for the lack of responses, I was away for a few weeks

> Please try https://github.com/openwrt/luci/pull/5833

I'm not terribly familiar with how to run the proposed change on my router. Is there a guide I can follow?

closed 2022-06-24 by aparcar

CAFxX, 2022-08-06:
I just reinstalled the current version without issues using the new functionality. Thanks @aparcar!

aparcar, 2022-08-06:
Excellent, spread the word

CAFxX, 2022-08-07:
Updated https://openwrt.org/faq/no_space_left_on_device to mention this