openwrt/luci #5843: Rebooting can lock you out of LUCI

QA-Manager, 2022-06-18:
With the 22.02.3 release, if you:
 - Make any settings change
 - Click 'Save'
 - DO NOT click 'Apply'
 - reboot
 
You will get a warning about unsaved changes, but the router will reboot. However, when LUCI comes back up it does not set a new cookie in your browser, and you won't be able to login. (This may not be 100% of the time, but if not 100% it is close to 100%.)

Deleting the cookie will fix the problem, but I think very few users will try that. This is a bad bug, despite the easy workaround, because the workaround is so extremely non-obvious. Most people will try cycling power on the router a few times, then they may try a factory reset and wipe out all of their work unless they have been diligent about making backups every few minutes.

I have 22.02.3 installed on a Linksys WRT1900-ACS, but I doubt that the hardware makes a difference.

pepe2k, 2022-06-18:
> This should move to openwrt/luci

Done!

Cheers,
Piotr

feckert, 2022-06-20:
https://github.com/openwrt/luci/issues/5843

commit e1932592 (2022-07-08): luci-base: use different cookie names for HTTP and HTTPS

Since HTTP cookies may not overwrite HTTPS ("secure") ones, users are
frequently unable to log into LuCI when a stale, "secure" `sysauth` cookie
is still present in the browser as it commonly happens after e.g. a
sysupgrade operation or when frequently jumping between HTTP and HTTPS
access.

Rework the dispatcher to set either a `sysauth_http` or `sysauth_https`
cookie, depending on the HTTPS state of the server connection and accept
both cookie names when verifying the session ID.

This allows users to log into a HTTP-only LuCI instanc [... cut]
commit 08fb3839 (2022-08-25): luci-base: use different cookie names for HTTP and HTTPS

Since HTTP cookies may not overwrite HTTPS ("secure") ones, users are
frequently unable to log into LuCI when a stale, "secure" `sysauth` cookie
is still present in the browser as it commonly happens after e.g. a
sysupgrade operation or when frequently jumping between HTTP and HTTPS
access.

Rework the dispatcher to set either a `sysauth_http` or `sysauth_https`
cookie, depending on the HTTPS state of the server connection and accept
both cookie names when verifying the session ID.

This allows users to log into a HTTP-only LuCI instanc [... cut]